ros-drivers/rosserial #157: Adjust queue size in rosserial-python for arduino

pando85, 2015-01-26:
I need to adjust queue size in my  arduino application, and I don't find nothing about it. 
I tried to review the code but it's so complicated to my python level.
I have some subscribers and I need to adjust queue size to 1.

mikepurvis, 2015-01-26:
Are you wanting it to be specifically control the queue size, or are you just trying to banish the rospy warning about queue_size being unset?

pando85, 2015-01-26:
I want to control the queue size of my subscribers in my arduino node.

mikepurvis, 2015-01-26:
Can you describe more about your project? Explicit queue size mostly comes into play with messages which are very large and/or very frequent. The normal use cases for rosserial don't typically need control this finely grained.

pando85, 2015-01-26:
I have built an hexapod based on arduino mega, and I'm integring it with ROS.
My arduino have some topics to receive diferents orders, mainly that orders are traducted to movements and I want that subscriptors in arduino just can handle the last order that they can handle.

Nowadays, arduino subscriptos are handling are orders that are published into the topics, and that produce that my robot still movement although you are not publishing topics.

Thanks for your time.

PaulBouchier, 2015-01-27:
This is an interesting question. If I publish 1000 messages and my consumer
takes one each second, are they queued up so that it will take 1000 seconds
before it runs out of messages?

This page has relevant information:
http://wiki.ros.org/roscpp/Overview/Publishers%20and%20Subscribers
It says you can set the queue size on the subscriber, and the oldest
message will be thrown away if the queue is full. IOW you have a choice of
dropping messages with a short queue, or keeping them with a long queue.

Perhaps the rosserial python should accept a parameter for queue size. Only
problem is, it would be global - likely undesirable. So thinking through
this, it does seem consistent with the rest of ROS that an arduino
rosserial subscriber should be able to set its queue size.

A hacky workaround would be to publish in ROS, and subscribe with a regular
ROS subscriber which sets its queue size to 1, and republishes on a topic
that the Arduino subscribes to:
Publisher -> topic1 (queue size 1) Subscriber -> Publisher -> topic 2
Arduino subscriber

Paul

pando85, 2015-01-27:
Interesting answer, but I have tried that, in fact, I did and intermedial node to try to minor the bug. But your suggestion still have a problem.

If you have in the middle:
Publisher -> topic1 (queue size 1) Subscriber -> Publisher -> topic 2
You are publishing one topic in the time that you process that or in a regular interval or even you can try to create and algoritm to modify the publication time.

But the problem is that you don't know the time of processing that message in your arduino, so you can't know the publication interval.

Example:
You publish 1000 messages in one second.
Your node filter that messages publishing just one per second.
Your arduino process first messages in 1.5 second, the second one in 1 second, the third in 1.2 second.
Your are almost synchronaise.

But if your arduino for example changes speed of processing that message you could get processings times of two or three second and you will have to change the publication time. 

In my case, it's not eassy to know the time of processing one message and it's complicate to develop one algoritm to readjust publication time in fuction of the differents parameters.

In my hexapod you can change height, foot distance, step size, gait, speed...

If you only have one message in queue it will perfect.

pando85, 2015-01-29:
Are there any way to do that? Could you give some clue to try to modify the code for my own?
I'm really interesting in do it work.

pando85, 2015-01-30:
Ok, I think that now I understand the magnitude of the problem.
rosserial python just take messages from topic and publish into arduino buffer so if you change rosserial subscriber queue size you dont get nothing more that you could lose some messages.

Now I'm thinking in the first solution becouse this is a problem that rosserial cant solve I think.

closed 2015-01-30 by pando85

Knightowl3128, 2019-12-02:
@pando85 I'm having this exact problem. Did you find a way out of this?

imanshahmari, 2022-06-16:
Same here problem here. I set the queue size to be 1 in arduino code but when connecting to it from my computer via rosserial it get the following error:

Tried to publish before configured, topic id 126