ros-drivers/rosserial #158: Warning failed to generate libraries for the following packages: ***

rummanwaqar, 2015-01-29:
Running `rosrun rosserial_client make_libraries .` inside the `sketchbook/libraries` generates the following error:

Warning failed to generate libraries for the following packages: ***
  geometry_msgs
  sensor_msgs
  shape_msgs
  my_msgs

After a lot of testing, I found that the problem only occured if I used arrays (e.g. `float32[3] xyz`). Replacing it with three different fields (e.g. `float32 x`, `float32 y`, `float32 z`) fixed the problem.

*How do I use arrays in my messages and still make sure they compile?*

mikeferguson, 2015-01-29:
@rummanwaqar I presume you meant to run rosserial_arduino or similar package's make_libraries -- but I just did a check out and tested and I can reproduce the error.

@mikepurvis this might be a major regression, previously the only packages that failed to generate were those with services with no return....

rummanwaqar, 2015-01-29:
@mikeferguson is there a previous version or something I can grab that works without this error?

mikepurvis, 2015-01-29:
Gracious, that's bad. Looking now.

The 0.6.3 release tag should be before the problem was introduced (almost certainly #152).

mikepurvis, 2015-03-11:
Fixed by #162.

closed 2015-03-11 by mikepurvis