ros-drivers/rosserial #159: Multicolor support

chuck-h, 2015-02-03:
I have been intending to write some mutidrop extensions to rosserial and might start on it soon. The idea is one serial port on the PC talking to a chain of small sensor/actuator devices. This might be RS485 tristate or daisy-chain token passing.

I would be happy to hear any comments on use cases or recommendations on implementation.

mikepurvis, 2015-02-03:
My only multi-drop experience is I2C and CAN. How does 485 handle bus contention?

chuck-h, 2015-02-03:
I have seen host-managed polling. In that case we might want to negotiate polling interval at topic setup time.

Alternatively a token-passing round robin protocol (i.e. the token is a "talking stick").

PetteriAimonen, 2019-09-28:
@chuck-h Did you ever get further with this?

I'm interested in using rosserial with RS485, though to start with I'll just have one device. I've been considering using host-managed polling, possibly with a way for the clients to signal they have data to send.

Something like this might work for an interrupt scheme:
- Host sends 0x00 before every packet
- If bus is idle for at least 2 byte times, then node can send their bus id as a single byte
- If host receives a byte, it will poll that node
- If the id byte happens to collide with the 0x00 preamble, it doesn't matter as the received value of preamble byte is not checked.
- If multiple id bytes collide, host might poll wrong node. Correct nodes can repeat their interrupts after random time if they still have data to send.

Common USB-RS485 converters have too much latency to do any collision detection on host side. But on device side it should be easy enough to detect start bit from the host before sending the interrupt byte.

Still needs extra fields to specify direction and bus id in the packet transfers.

arlenn, 2019-10-01:
I am interested to know what you come up with...  I have a project in-mind to use multi-dropped two-wire RS485; I was hoping Rosserial might already do this.