ros-drivers/rosserial #160: ROSserial arduino and I2C communication

nvoltex, 2015-02-05:
I have been trying to get ROSserial to work in combination with an I2C communication with an IMU, however it seems like the ROSserial communications are messing the I2C communication/DMP initialization (or maybe the other way around.).

I'm using an MPU9150 and for starters I only wanted to publish the yaw, roll and pitch to a ROS topic. Using this library (https://github.com/rpicopter/ArduinoMotionSensorExample) I was able to achieve that result. 
However when I tried to add an subscriber to the arduino node, I start getting and error related to some problem on the DMP and can't get the desired information from the IMU. 

Since then I have tried other libraries and end up with similar problems. In fact using the implementation from one of the examples of the I2Cdev library I'm able to get the IMU data as long as I don't add ROSserial to the skectch. Once I do it, it seems like there is some problem with the DMP configuration.

Is this problem common? Can you help me out?
Thanks in advance!

tlalexander, 2015-09-16:
Hello!
I've messed with those IMUs a little bit. I know in the sketches I was using, there is an interrupt to read the MPU that fires when the data ready line activates. Is it possible that the data ready line is activating at the same time that ROSserial is trying to read messages? They may be both trying to use the CPU at the same time. If you have access to an oscilloscope, try toggling some pins when each function runs, and see if they are colliding. You may need to add some flow control. For example, in the IMU interrupt instead of doing the I2C read there, just set a "data ready" flag and then read the data in the main loop after the nh.spinOnce(); function.

It doesn't look to me like ROSserial is using interrupts, so if you don't access the I2C bus in an interrupt either, you should be able to get everything to play nicely.

asukiaaa, 2019-04-05:
I succeeded in publishing data of an i2c device with using Arduino Mega2560 but failed with using Uno and Leonardo.
Is there any way to support i2c and ros together on Uno or Leonardo?

asukiaaa, 2019-04-10:
Sorry, I succeeded in publishing and subscribing ros topic with using i2c lcd on Uno and Leonardo.
Maybe my program have some problem.

hariharan382, 2022-03-01:
but how ,please explain the solution