ros-drivers/rosserial #163: ROS Jade

mikepurvis, 2015-02-13:
ROS Jade is another chance to rev the wire protocol, especially for the sake of #130, which would enable more uses of rosserial, such as tunnelling larger messages such as images and pointclouds between ROS and other systems where it is impossible or undesirable to run roscpp.

However, I'm not able to commit time to it apart from doing the regular release.

@ibaranov-cp, @cjwills, can one of you guys get together a PR for this? We should have a release into Jade no later than Feb 28th IMO, so that leaves a few weeks for implementation and review.

ibaranov-cp, 2015-02-13:
I can certainly make a priority to get an initial release into our github
before then. Looking into it now

mikepurvis, 2015-04-01:
Well, so much for a Feb 28th release to Jade. I don't think we can wait any longer. Going to release to Jade this week with Indigo's wire protocol.

closed 2015-04-23 by mikepurvis