ros-drivers/rosserial #171: packet protocol version flag in hydo/indigo is 0xfe or 0xfd?

bmagyar, 2015-03-08:
According to the 'Packet format' section in the documentation it is 0xfd
http://wiki.ros.org/rosserial/Overview/Protocol

According to the code is it 0xfe (and I prefer to believe this :))
https://github.com/ros-drivers/rosserial/blob/indigo-devel/rosserial_server/include/rosserial_server/session.h#L135

Also, my Tiva says 0xfe. 

Please let me know if I overlooked something here, if not the documentation needs a tiny update.

Note that this may also affect the protocol version flag of Jade which is mentioned on #130

mikepurvis, 2015-03-10:
I agree with your assessment— whoops. I'm in the process of trying to put together the Jade release right now, so I can clean up the docs as part of that. Will leave this ticket open until that has been completed.

mikepurvis, 2015-03-11:
Docs have been updated. Thanks again for the heads-up.

closed 2015-03-11 by mikepurvis

bmagyar, 2015-03-11:
Nice! Thanks for the fast response!