ros-drivers/rosserial #172: rosserial_server SerialSession destructor is private

mitchellwills, 2015-03-10:
Is there a reason why the SerialSession destructor is private? This prevents it from being cleaned up properly when the program shuts down. The serial_node simply allocates a new one on the heap and throws away the pointer to it. Is this a mistake or is there a threading reason for it?

mikepurvis, 2015-03-10:
Sounds like a mistake. I would welcome a fix to this.

mitchellwills, 2015-03-10:
In digging into this further I am very confused by the expected lifecycle of SerialSession and Sessions in general. The session class seems to want to manage it's own lifecycle and delete itself when the connection fails and rely on the destructor to spawn a new instance of itself. This seems like a bad idea as it gives the user no control over the actual object and forces it to be allocated on the heap (in addition to causing a number of other issues: http://isocpp.org/wiki/faq/freestore-mgmt#delete-this)

mitchellwills, 2015-03-10:
This is also a memory leak

mikepurvis, 2015-03-11:
I agree that this architecture is non-ideal. In truth, it's a leftover from rosserial_server's origin as the boost [async_tcp_echo_server](http://www.boost.org/doc/libs/1_55_0/doc/html/boost_asio/example/cpp03/echo/async_tcp_echo_server.cpp) example, which uses this basic approach. The memory leak is extremely slow, so I'm not terribly concerned on that front, but it is still an indication of poor design.

I would be delighted to entertain proposals for how the architecture of rosserial_server might be rationalized and documented.

mitchellwills, 2015-03-11:
Wow i'm really surprised that's in an example. I've done a few projects recently that have been based around boost asio so I may take a look at potential ways to clean it up if your open to that. The biggest issue with the current architecture is that it gives you no way to interact with the session once it's created (explicitly shut it down, query state, etc)

mikepurvis, 2015-03-11:
Yes, I agree, and that's definitely a flaw. I suppose the best would be some kind of data structure which manages the session pointers, perhaps a vector or map, though it's not clear what you'd key them on. The main thing is being able to iterate them, and having a way to move that tortured session restart logic from within the session to an external manager process (ie, the session has a way to signal to the manager "shut me down now" or "shut me down and reopen me").

In any case, yes, please think on it and we'll discuss further soon.

mitchellwills, 2015-03-11:
I would initially propose something similar to [async_web_server_cpp](https://github.com/WPI-RAIL/async_web_server_cpp) where there is a server like object that creates sessions (like HTTPConnection) that handles the actual session.