ros-drivers/rosserial #175: rosserial_server service support

attermann, 2015-03-11:
I understand that development of service support is already underway in rosserial_server.  As it currently isn't working for me in either indigo release or devel, I wanted to find out what the status is before I delve in deeper.  I'm happy to help if it still needs more work.

For reference, I'm seeing warnings in rosserial_server like the following when running the ServiceServer example on Arduino...

[ WARN] [1426091785.412954964]: Received message with unrecognized topicId (2).
[ INFO] [1426091785.420138438]: waitForService: Service [/message_info] has not been advertised, waiting...
[ WARN] [1426091790.441098205]: Timed out waiting for message_info service to become available.

I assume that topicId (2) has something to do with the service advertised from Arduino.  I also noticed that rosserial_python doesn't complain about message_info service not being available, so is it safe to assume that Arduino is advertising this service properly, and it will work in rosserial_server once services are fully implemented?

mitchellwills, 2015-03-11:
This has nothing to do with a service advertised from the arduino. You also need to run rosserial_python/message_info_service.py when running rosserial_server
See: https://github.com/ros-drivers/rosserial/blob/indigo-devel/rosserial_server/launch/serial.launch
and http://wiki.ros.org/rosserial_server#Overview

attermann, 2015-03-11:
Thanks, I just came across that example launch file myself this morning.  It would be nice to not have any dependence on rosserial_python however.  And this leads back to my original question; what is the status of service implementation in rosserial_server?

mikepurvis, 2015-03-11:
I agree. One option would be to supply a list of message definitions to compile in, so that common message use cases wouldn't require the shim. If it were an extensible mechanism, platform-specific users could add in their own message definitions as well.

mitchellwills, 2015-03-11:
The issue with that is it requires rosserial_server to be built from source to use a new message type

attermann, 2015-03-11:
I am new to rosserial, and to ROS for that matter, so I don't have a very good understanding of message_info and service_info.  rosserial_python must be getting these definitions from somewhere at runtime though, so it doesn't seem too far fetched for rosserial-server to be able to retrieve them at runtime as well, perhaps by parsing msg and srv files into some internal structure. Could this work, or am I way off?

henoSH, 2016-03-15:
hello i tried also to use ServiceClient but i got the same error : waitForService: Service [/message_info] has not been advertised, waiting... 
have someone find a solution?

attermann, 2016-03-15:
Ensure that rosserial_python/message_info_service.py is included in your launch file.  This python "shim" is necessary when using rosserial_server.

henoSH, 2016-03-16:
yes it is already included 
<launch>
  <node pkg="rosserial_server" type="socket_node" name="rosserial_server" />
  <node pkg="rosserial_python" type="message_info_service.py" name="rosserial_message_info" />
</launch>