ros-drivers/rosserial #177: sync issues when usb disconnects and reconnects

mgerdzhev, 2015-03-23:
Running rosserial_python after uploading and arduino program successfully syncs up and works perfectly. However if I unplug the usb and then plug it back in, running rosserial_python is unable to sync with the device. Reuploading the program and starting rosserial_python fixes the issue again. The problem has been previously reported here: http://answers.ros.org/question/29378/rosserial-and-arduino-resets/ and one of the comments proposes a fix to SerialClient.py where it sets the baud rate to 0 on startup for a few seconds. That change seems to fix the problem for me. Is it possible to incorporate it into the main tree? This problem happens to me on 3 different computers and 2 different arduino Mega.

tonybaltovski, 2015-03-24:
@mgerdzhev I believe this is a hardware issue since Arduino resets itself when it connects via serial.

Tones29, 2016-06-22:
Since a few weeks, I have exactly the same problem with different types of devices (Arduino Uno, Arduino Micro). I tried several sketches (HelloWorld-Example, own sketches). I am using ros indigo with ubuntu 14.04, Kernel 3.16.0-73.

vmatos, 2016-06-22:
@mgerdzhev I think this has been a issue with pyserial, Ubuntu 14.04 and its kernels.
I usually also have that problem. My workaround is simply running `rosserial_python` with a dummy baud, exit, and run it again with the correct baud. 

Therefore I believe the workaround presented on the ROS answers would work. However it's ugly and specific for a problem which affects only people running 14.04. (right?)
Today I'm testing it on 16.04 to see if the problem continues.

Tones29, 2016-06-23:
@vmatos you are right, this seems to be a Kernel problem. I could reproduce the issue with the following Kernels (ubuntu 14.04, ROS indigo):
- 3.10.96
- 3.13.0
- 3.16.0

Luckily, I found three possible workarounds:
- Use the current xenial kernel with ubuntu 14.04 (version 4.4.0)
- Use your suggested procedure with the dummy baud rate 0
- Modify `SerialClient.py` as kvogt suggests [here](http://answers.ros.org/question/29378/rosserial-and-arduino-resets/).

None of the workarounds is really satisfying for me. There should be a lot of users out there, running into the same problem.

SouLeo, 2018-06-04:
Hi All, 

I'm running on Ubuntu 16.04 kernel: 4.13.0-43-generic, and I'm persistently (but at unpredictable times) getting the mismatch error message. I'm not sure if this has been your experience @vmatos after testing on 16.04, so I wanted to follow up with you.

anuja2201, 2022-01-25:
Hi all I am having the same issue of mismatch on STM blackpill. I have tried changing baud rate and other mentioned solutions as well. Is there anything that need to be set/cleared for blackipll to connect with rosserial.

matthewryanburgin191, 2022-04-02:
> Hi all I am having the same issue of mismatch on STM blackpill. I have tried changing baud rate and other mentioned solutions as well. Is there anything that need to be set/cleared for blackipll to connect with rosserial. 

Did you solve this in the end?

Thomas-DeClercq, 2024-10-04:
I'm experiencing the same issue on Noetic. However the fix proposed in the link doesn't seem to solve the issue