ros-drivers/rosserial #179: xbee_network.py error

rmolin88, 2015-04-01:
Hello, the problem that I am having is that I am able to successfully set up xbee with `rosrun rosserial_xbee setup_xbee.py` nodes. I have two Pro series 1 xbee's.
but then when I try for one of xbee's communicate with `rosrun rosserial_xbee  xbee_network.py /dev/ttyUSB0 0` I get:
```
RosSerial Xbee Network
Contacting Xbees :  ['\x00\x00']
/opt/ros/indigo/lib/python2.7/dist-packages/rosserial_python/SerialClient.py:336: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  self.pub_diagnostics = rospy.Publisher('/diagnostics', diagnostic_msgs.msg.DiagnosticArray)
Exception in thread Thread-7:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosserial_python/SerialClient.py", line 434, in run
    if self.port.inWaiting() < 1:
AttributeError: FakeSerial instance has no attribute 'inWaiting'
```
I have joy_node publishing messages on the other end.
I am running ros-indigo on an Odroid U3. I get the same error on my PC running ros-hydro when I try to contact the xbee on the Odroid (I know this is useless at the moment because the Odroid xbee is not transmitting ). 
This problem has also being reported [here](http://answers.ros.org/question/187443/xbee-network-error/) and [here](http://answers.ros.org/question/206358/raspbian-ros-indigo-rosserial_xbee-error/) most recently.
Any help will be much appreciated.
Thank you.

mikepurvis, 2015-04-01:
Thanks for this report. The bug would have been introduced in 370f5791da88ef930d6c1e035118bd07102b2dd5.

The [FakeSerial class here](https://github.com/ros-drivers/rosserial/blob/indigo-devel/rosserial_xbee/scripts/xbee_network.py#L55) requires an inWaiting method be added. I would welcome a PR for this.

lucasplus, 2015-07-24:
@rmolin88 If the pull request works for you, I think you can close this issue. I'm not sure. I've never opened an issue.

rmolin88, 2015-07-24:
Ohh I am so sorry. I had never open and issue before. I tested and now I get another error but I dont think is related to this one. Thank you Lucas for your help!

closed 2015-07-24 by rmolin88

lucasplus, 2015-07-24:
@rmolin88 What error did you get? Maybe I've seen it and know what is happening.

rmolin88, 2015-07-24:
@lucasplus this is the error I am getting
`Creation of publisher failed: unpack requires a string argument of length 2`
I believe it has to do that I brute force the new 2 lines of code into the current ros lib
I want to try cloning the packet into my ws, fixing the issue there see if that fix its.
Correct me if I am going in the wrong direction. Just have not had the chance to try it.
Thank you again Lucas

lucasplus, 2015-07-24:
@rmolin88 Sorry. I never have seen that one. So it seems like you are working with a repository that makes you unsure that you have the correct changes. As a sanity check you can install rosserial from source in a completely different work space. I did that this morning on my box. You just to follow the instructions from [this rosserial_arduino tutorial](http://wiki.ros.org/rosserial_arduino/Tutorials/Arduino%20IDE%20Setup) "Installing from Source onto the ROS workstation" Was the rosserial on your box installed with apt-get or did you install it from source?

rmolin88, 2015-07-25:
@lucasplus Hey I tried your suggestion still getting same error. I had installed rosserial-xbee through apt-get. I removed that through apt-get remove. Followed the tutorial but still getting same error. Going to look at the code see why it launches that error.
Maybe I am doing it the wrong way. All I am doing is just on my pc I launch `rosrun rosserial_xbee xbee_network /dev/ttyUSB 0` and then on my box same thing but 1. Where my box has the coordinator.

lucasplus, 2015-07-28:
@rmolin88 I see. A more complex problem than I thought. Also, your use application is different than mine. I using mine to talk to an Arduino. You are using yours to talk between PCs. One thing that comes to mind, when I used `sudo apt-get remove`, I had to type `sudo apt-get remove ros-indigo-rosserial` and then hit TAB which auto completed packages that started with "ros-indigo-rosserial" I had like like 5 or 6. Meaning you may not of completely removed all packages attributed with rosserial. I think at this point you should created a post on ROS answers describing your setup and steps you took in detail. Post the link here and I'll check it out.

ButlerRobotics, 2015-08-18:
I've been getting the same AttributeError issue. If I try building from source I get this error instead:

`[ERROR] [WallTime: 1439899757.549239] Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino`

I have tried various branches just to see if this would fix it but no joy. 
Xbee was successfully set up but cannot connect. 
The XBees work on arduino serial monitor and I can get rosserial to work but not rosserial_xbee.
Any ideas?

ButlerRobotics, 2015-08-19:
I realise this new error could be unrelated to the original error but thought I would log it here just in case there was some relevance. I can log a new issue if that is preferred?

lucasplus, 2015-08-19:
@ButlerRobotics Ok. I believe the errors are unrelated. The error from the issue on this page was simply because a method wasn't defined in the FakeSerial class. It works when you build from source because the maintainer probably hasn't cut a new release. [see here](https://github.com/ros-drivers/rosserial/pull/190#issuecomment-124297660) 

I have got the "Unable to sync with device" error before. It isn't something to log a new issue about yet. The error can come from for a few different reasons. Many times from hardware not being configured correctly. Have you followed [this tutorial?](http://wiki.ros.org/rosserial_xbee/Tutorials/Example%20Network) Especially the part about "Configuring the XBees" What I think is the right thing to do, if you follow the tutorial and still get an error, is make a post on ROS Answers. Describe your system in detail. What worked and what didn't. Basically, steps to reproduce the error. You can post the link here and I will take a look at it.

ButlerRobotics, 2015-08-19:
@lucasplus Great, thanks for your help! I have already done the steps you suggested so I will post on ROS answers and share the link here.

lucasplus, 2015-08-19:
Great.

ButlerRobotics, 2015-08-20:
@lucasplus Hey my post finally passed moderation on ROS Answers so it's up there now. Would appreciate any help. This is turning out to be quite an unexpected blocker for me. Thanks.

http://answers.ros.org/question/216191/rosserial_xbee-unable-to-sync-with-device/

ButlerRobotics, 2015-08-26:
@lucasplus any ideas about this?

lucasplus, 2015-08-26:
@ButlerRobotics I was traveling the end of last week and this week is quite hectic. I looked at your question on ROS Answers and nothing came to mind. I might be able to look more in depth next week.

ButlerRobotics, 2015-08-26:
@lucasplus thanks for having a look. Really appreciate it.

lucasplus, 2015-09-13:
@ButlerRobotics The project that I was working on (which used the XBees) ended. I looked at your question and didn't have anything that came to mind. The only thing I could recommend is put debug statements in your python code to figure out exactly what line it is failing on.

ButlerRobotics, 2015-09-13:
@lucasplus thanks for having a look. I give that a go. One last question: what version of ROS and Ubuntu were you using for this project?

lucasplus, 2015-09-14:
@ButlerRobotics Ubuntu 14.04 and ROS Indigo. I installed rosserial from source using the indigo-devel branch.

ButlerRobotics, 2015-09-14:
@lucasplus great thanks.