ros-drivers/rosserial #181: Performance issue of rosserial_arduino

ymkim92, 2015-04-03:
Hi,

I have used rosserial_arduino to connect Arduino to my roscore on Linux.
When I run roscore on my laptop. The performance is quite stable.
I am sending data at 100Hz rate in both direction.
But, on the embedded Linux of Gumstix overo, its performance is very unstable.
It is changing in range of 80Hz ~ 100Hz and sometimes the connection is broken with this error:

    [ERROR] [WallTime: 1428040762.537858] Lost sync with device, restarting...

Can I fix this problem?

mikepurvis, 2015-04-06:
I don't have a lot of experience trying to run the ROS side on resource-constrained platforms, but I would suggest trying [rosserial_server](http://wiki.ros.org/rosserial_server#Usage) in place of rosserial_python. It's a C++ implementation of the server, and more performance-oriented.

ymkim92, 2015-04-08:
Thanks a lot. It is working well in my laptop. Then, I tried to run rosserial_server on my Gumstix. But, they don't support rosserial_server. I think it is still unstable and in development.

mikepurvis, 2015-04-08:
Hmm, is it just that rosserial_server isn't available in the armhf debians? If you tried to compile it yourself and ran into an issue, I'd be interested to have that reported so it can be addressed.

ymkim92, 2015-04-10:
I compiled rosserial_server and install it in Gumstix. The performance is much more stable. Great!
It is almost the same as on my laptop with rosserial_python. But still I have a problem. That is both ROS and Arduino did not make stable timing in 100Hz.

closed 2015-04-12 by ymkim92