ros-drivers/rosserial #182: action support?

v01d, 2015-05-16:
Hi, I was wondering if rosserial supports actions? I'm considering using rosserial to implement the controller of a robotic manipulator, which is programmed in an Arduino Mega, by directly offering it as a FollowJointTrajectory Action server to a ROS PC. However, if actions are not supported I would need to solve this otherwise.

mikepurvis, 2015-05-19:
Not directly, but under the covers an action is just a handful of topics, so you can implement an action client or server manually without much difficulty (I have done this).

closed 2015-05-25 by mikepurvis

eborghi10, 2017-04-16:
@mikepurvis Do you have a Github repository where we can see how do you implement it?
That would save us a lot of time.
Thanks in advance.

mikepurvis, 2017-04-17:
Sorry, nothing public. If you're getting stuck somewhere particular, feel free to open a new issue to discuss it.