ros-drivers/rosserial #184: Error loading multiple params

rummanwaqar, 2015-05-21:
Sometimes when I am loading multiple params from a launch file, some of them don't get loaded and the default value is used. Restarting the node sometimes fixes it. However I am trying to find the source of the problem and trying to see if I can do something to fix it and make it more reliable. I am using a _teensy 3.1_ for development.

**Teensy part**
```
while( !nh.connected() ){
    nh.spinOnce();
  }

//get parameters and assign pins
  int horPins [4];
  if(!nh.getParam("~horPins",horPins,4)) {
    horPins[0] = HOR_DIR_L;
    horPins[1] = HOR_PWM_L;
    horPins[2] = HOR_DIR_R;
    horPins[3] = HOR_PWM_R;
    nh.logwarn("motor: loading default values for horizontal pins;");
  } else {
    nh.loginfo("motor: loaded params for horizontal pins;");
  }

  int verPins [4];
  if(!nh.getParam("~verPins",verPins,4)) {
    verPins[0] = VER_DIR_L;
    verPins[1] = VER_PWM_L;
    verPins[2] = VER_DIR_R;
    verPins[3] = VER_PWM_R;
    nh.logwarn("motor: loading default values for vertical pins;");
  } else {
    nh.loginfo("motor: loaded params for vertical pins;");
  }

  int horFlip[2];
  if(!nh.getParam("~horFlip",horFlip,2)) {
    horFlip[0] = 0;
    horFlip[1] = 0;
    nh.logwarn("motor: loading default values for horizontal flip;");
  } else {
    nh.loginfo("motor: loaded params for horizontal flip;");
  }
  int verFlip[2];
  if(!nh.getParam("~verFlip",verFlip,2)) {
    verFlip[0] = 0;
    verFlip[1] = 0;
    nh.logwarn("motor: loading default values for vertical flip;");
  } else {
    nh.loginfo("motor: loaded params for vertical flip;");
  }
  int minThrust;
  if(!nh.getParam("~minThrust",&minThrust)){
    minThrust = MIN_THRUST;
    nh.logwarn("motor: loading default value for min thrust;");
  }
  else{
    nh.loginfo("motor: loaded params for min thrust;");
  }
```

**Launch file**
```
<launch>
	<node
		pkg="rosserial_python"
		type="serial_node.py"
		name="motor"
		output="screen"
		respawn="true">

		<!-- Parameter Set -->
		<rosparam file="$(find arvp_main)/params/motorParam.yaml" />
		
	</node>
</launch>
```

**Param file**
```
# usb info
port: /dev/arvp/motor 
baud: 57600

# min thrust
minThrust: 10

#pin assignments
horPins:
- 16 #dirL
- 20 #pwmL
- 17 #dirR
- 21 #pwmR

verPins:
- 18 #dirL
- 22 #pwmL
- 19 #dirR
- 23 #pwmR

#flip
horFlip:
- 0 #L
- 0 #R

verFlip:
- 0 #L
- 0 #R
```

I have a quick deadline coming up so a quick solution or any help so I can fix it myself would be really appreciated :+1:

mikepurvis, 2015-05-21:
Ultimately, rosserial has no sequence numbers or acks/nacks— so if you get a bit flip in either direction, the checksum will fail and you'll lose your parameter. This is unfortunate; from the client's perspective, there's no way to differentiate between a missing parameter and a bit error, timeout, etc. (This is most of the reason I haven't implement parameters in rosserial_server.)

That said, there are ways you could be a lot more sure. For example, if your firmware knows for sure that the parameters will be present, you could simply loop around the `getParam` call until you get success rather than falling back to a default value.

Without a protocol change, I don't think there's a way to be completely sure, but certainly the current implementation in rosserial_client could be a lot more enlightened (for example, making the request multiple times instead of just once).

Hope that helps.

rummanwaqar, 2015-05-21:
Thanks a lot. I will just loop around with some timeout. :+1:

rummanwaqar, 2015-05-21:
I am now using a loop to try reading the value at least 5 times, otherwise timeout
```
  count = 0;
  int verPins [4];
  while(!nh.getParam("~verPins", verPins, 4) && count < PARAM_TRY) { count++; }
  if(count < PARAM_TRY)
  // param was loaded
  {
    nh.loginfo("motor: loaded params for vertical pins;");
  } else
  // param timed out so use default values
  {
    verPins[0] = VER_DIR_L;
    verPins[1] = VER_PWM_L;
    verPins[2] = VER_DIR_R;
    verPins[3] = VER_PWM_R;
    nh.logwarn("motor: loading default values for vertical pins;");
  }
```

However after a lot of testing, I found that if I change a single param in the param file, the first time it always loads default values without even trying to load the params from the param file. Next time it properly loads the parameters.

mikepurvis, 2015-05-21:
Can you expand on "without even trying"? Is this with resetting the rosserial python server? In between runs, are you resetting or depowering your embedded board? Is it serial or USB comms?

rummanwaqar, 2015-05-21:
So to change params: 
* I turn off the rosserial node. (The microcontroller is programmed to restart when the node disconnects) 
* Change the value on the param in the param file.
* Restart the node.

The following only happens the first time the param file is changed.

Normally if the node cannot load the params it gives the error log saying unable to load params which I think originates from rosserial python server. In this case however I get no such error. Instead it only gives the warning I publish from my program when loading default values. `nh.logwarn("motor: loading default values for vertical pins;");`

mikepurvis, 2015-05-22:
Can you try inserting a 1 second pause (with rosserial spins in it) between the initNode and first getParam? I'm wondering if the first request message is getting lost in the initialization.

rummanwaqar, 2015-06-02:
I was looking at my code, and here is the general overview

nh.initNode();

//wait until conneded with rosserial before coninuing
while( !nh.connected() ){
  nh.spinOnce();
}

// makes calls to get all the parameters
getParam();

So basically after initiating the node, I generally wait for a couple of seconds until the node is actually connected. Do you still want me try it with a 1 second pause?

mikepurvis, 2015-06-05:
What you have should actually be okay— you're not sending any data until the topic setup messages have gone out, and the server will definitely pick those up. Not sure what else to suggest at this point.

closed 2018-02-20 by rummanwaqar