ros-drivers/rosserial #185: Serial::setTimeout should accept param by const reference

theidealist, 2015-06-02:
It's misleading (and inconsistent with the constructor version of the same API) to accept this parameter by  non-const reference as it indicates to me the user that it is being "captured" by reference and that I should therefore keep the timeout object I provide it around for at least as long as I have the Serial object itself, or that the object will be modified by the function itself.  Neither of these do I believe to be true after closer inspection of the code.

If there is data being changed in Timeout (I followed the pimpl implementation through to unix.cc:702) it should be advertised more clearly in the documentation.

Thanks!

mikepurvis, 2015-06-03:
rosserial and serial are unrelated projects. Please see [serial's bugtracker](https://github.com/wjwwood/serial/issues).

closed 2015-06-03 by mikepurvis

theidealist, 2015-06-03:
Okay, will report there.  Thanks!