ros-drivers/rosserial #186: Serial port read failure

mehtajaghvi, 2015-06-03:
I am using ultrasound example provided in ros_lib.
I am running that using rosrun rosserial_python serial_node.py at 57600 baud rate. This results in serial port read failure. 

[WARN] [WallTime: 1402665687.484972] Serial Port read returned short (expected 1 bytes, received 0 instead).
[WARN] [WallTime: 1402665687.485703] Serial Port read failure

I have tried using rosserial from git, from binary, changed baud rates, changed ports, re-installed ros_lib.
The last edit I did was change the timeout from 5 to 10 in serilclient.py, this does reduce the frequency of error.

miguel7676, 2015-07-26:
Hi there, have you had any luck sorting this issue out?

mehtajaghvi, 2015-07-29:
No, I could not find any answers. It does not affect the system as it is more of a warning but this happens with not just one sensor but multiple different kinds. I tried everything possible over internet but to no avail

miguel7676, 2015-07-29:
Oh ok thank you for the reply. Ive also searched the Internet and found nothing. Just some hints that it is the arduino buffers that are too full.