ros-drivers/rosserial #188: rosserial lost with device arduino

dmngu9, 2015-06-29:
i know this issue happened in the past before but seems like no appropriate answers are posted yet.
i am using mega board with ros. i have lost sync error sometimes. I managed to make it works somehow few weeks ago. But now problem suddenly appears again

can anyone please post any solution for this issue? should i use different microcontroler?

kk6axq, 2015-08-11:
I have the same issue. I documented it here: http://answers.ros.org/question/215687/rosserial-arduino-connectivity-problem/

kk6axq, 2015-08-12:
Are you using the Servo.h library in your Arduino code? I saw somewhere that that could be the cause of the problem but it didn't seem to work for me. It might work for you...

dmngu9, 2015-08-12:
i used servo library. I noticed that if my servo changed direction suddenly from 0 to 180 and vice versa repeatedly. lost sync occurs. Its fairly weird

kk6axq, 2015-08-13:
I get something similar but I found that if I restart the computer that
rosserial is running on it works. Do any of the devs know why this could be
happening.