ros-drivers/rosserial #189: catkin tools problem

niosus, 2015-07-20:
Hello!

Thanks for all the effort, the package you implement is a real life saver! 

This may be not in priority, but for those of us who use catkin_tools and therefore the command `catkin build <some_pkg>` is not working properly as it cannot find the appropriate targets.

Following the tutorials [here](http://wiki.ros.org/rosserial_arduino/Tutorials/CMake) there is no way (as of now) to use `catkin build` instead of `catkin_make`. That would be great as I use the `catkin_tools` for all my projects and it is mutually incompatible with `catkin_make` way.

niosus, 2015-07-20:
Oh, my bad, there was an error in rosserial_arduino tutorials and everything else is working just fine with catkin tools. Closing issue. Sorry for the noise.

closed 2015-07-20 by niosus

mikepurvis, 2015-07-20:
No sweat. Please consider fixing the tutorial, or at least posting what the issue was so that we can fix it. :)

skohlbr, 2016-09-01:
I'd actually be interested, as I think I ran into this problem, too (catkin build no finding a target that catkin_make appears to find).