ros-drivers/rosserial #192: Publish & Subscribe

sergiosrl, 2015-08-19:
Hi, I'm publishing an Array, and also using subscribe. The problem is I also want to be able to use Serial.read() and Serial.write(), and I'm not able to communicate probably because the serial is being used with the Subscribe. 

Is there any way to stop the subscribe or kill the node?

Thanks

mikepurvis, 2015-08-19:
Rosserial is not designed to share the serial connection with other traffic. There are ways it could be made to work, but the intention is that it has exclusive use of the serial connection.

What are you trying to do? Would your needs be met by using a device like the atmega32u4, which has the USB connection in addition to a UART? Then you can run rosserial on the USB connection, and still use the UART for your auxiliary serial port.

sergiosrl, 2015-08-21:
I used a USB Serial Light Adapter, and now I'm able to communicate like I wanted. Thank's

closed 2015-08-21 by sergiosrl