ros-drivers/rosserial #195: Serial Port read failure: Happens after using it a PC a few times

wkentaro, 2015-10-12:
```
% rosrun rosserial_python serial_node.py _baud:=115200 _port:=/dev/ttyUSB0
[INFO] [WallTime: 1444637692.706220] ROS Serial Python Node
[INFO] [WallTime: 1444637692.713405] Connecting to /dev/ttyUSB0 at 115200 baud
[WARN] [WallTime: 1444637694.821324] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637694.822240] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637694.822828] msg len is 8
[WARN] [WallTime: 1444637694.829586] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[WARN] [WallTime: 1444637694.838124] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637694.838809] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637694.839316] msg len is 8
[WARN] [WallTime: 1444637694.848570] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637694.849394] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637694.850085] msg len is 8
[WARN] [WallTime: 1444637694.855982] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[WARN] [WallTime: 1444637694.865405] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[WARN] [WallTime: 1444637694.873120] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[WARN] [WallTime: 1444637694.875371] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637694.875875] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637694.876383] msg len is 8
[WARN] [WallTime: 1444637694.883806] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637694.884618] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637694.885361] msg len is 8
[WARN] [WallTime: 1444637694.891931] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[WARN] [WallTime: 1444637694.901054] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[WARN] [WallTime: 1444637694.904514] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637694.905113] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637694.905661] msg len is 7
[WARN] [WallTime: 1444637694.913156] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637694.913758] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637694.914236] msg len is 7
[WARN] [WallTime: 1444637694.921501] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[WARN] [WallTime: 1444637694.932071] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637694.932646] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637694.933200] msg len is 8
[WARN] [WallTime: 1444637694.940256] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637694.940928] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637694.941415] msg len is 8
[WARN] [WallTime: 1444637694.948994] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637694.949656] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637694.950239] msg len is 8
[WARN] [WallTime: 1444637694.957406] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637694.957985] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637694.958421] msg len is 8
[WARN] [WallTime: 1444637694.967479] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637694.967981] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637694.968446] msg len is 8
[WARN] [WallTime: 1444637694.976657] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637694.977377] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637694.977858] msg len is 8
[WARN] [WallTime: 1444637694.986290] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637694.987060] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637694.987538] msg len is 8
[WARN] [WallTime: 1444637694.994574] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637694.995213] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637694.995628] msg len is 8
[WARN] [WallTime: 1444637695.002952] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444637695.003566] Packet Failed :  Failed to read msg data
[INFO] [WallTime: 1444637695.003976] msg len is 8
[ERROR] [WallTime: 1444637695.011956] Tried to publish before configured, topic id 101
[ERROR] [WallTime: 1444637695.013795] Tried to publish before configured, topic id 102
```

Do you have any idea to solve this?

Firstly I wonder the cause is the versions of dependencies of rosserial_python, but I found that there are no differences between them.

wkentaro, 2015-10-13:
At other time, I have error below:
```
% rosrun rosserial_python serial_node.py _port:=/dev/ttyUSB0 _baud:=115200
[INFO] [WallTime: 1444743760.324098] ROS Serial Python Node
[INFO] [WallTime: 1444743760.331972] Connecting to /dev/ttyUSB0 at 115200 baud
[WARN] [WallTime: 1444743762.445800] Serial Port read failure: device reports readiness to read but returned no data (device disconnected?)
[INFO] [WallTime: 1444743762.446595] Packet Failed :  Failed to read msg data
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/rosserial_python/serial_node.py", line 82, in <module>
    client.run()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosserial_python/SerialClient.py", line 475, in run
    rospy.loginfo("msg len is %d",len(msg))
UnboundLocalError: local variable 'msg' referenced before assignment
%
```
the environement I use rosserial is:
* Ubuntu 14.04
* indigo
```
% for i in $(rospack depends rosserial_python); do
echo $i $(rosversion $i)
done
genmsg 0.5.6
genpy 0.5.5
cpp_common 0.5.6
rostime 0.5.6
roscpp_traits 0.5.6
roscpp_serialization 0.5.6
message_runtime 0.4.12
catkin 0.6.15
gencpp 0.5.3
genlisp 0.4.15
message_generation 0.2.10
rosbuild 1.11.9
rosconsole 1.11.14
std_msgs 0.5.9
rosgraph_msgs 1.11.1
xmlrpcpp 1.11.14
roscpp 1.11.14
rosgraph 1.11.14
rospack 2.2.5
roslib 1.11.9
rospy 1.11.14
rosserial_msgs 0.6.4
diagnostic_msgs 1.11.8
```
but in other PC whose environment is same as above, serial_node.py does work properly.

vmatos, 2015-10-19:
I just had the same result. And it was working on this computer a few weeks before.
I seems unrelated to rosserial package: [Ubuntu kernel bug ticket](https://bugs.launchpad.net/ubuntu/+source/python2.7/+bug/1501240)

Downgrade to 3.13.0-62-generic and it should work.

mikepurvis, 2015-10-19:
@vmatos Thanks for the info. Sounds like rolling back (or forward?) the kernel version or switching to `rosserial_server` corrects the issue.

closed 2015-10-19 by mikepurvis

wkentaro, 2015-10-20:
@vmatos Thanks.
I upgraded the kernel version and the issue is fixed.