ros-drivers/rosserial #199: [Arduino] make_libraries: has no attribute _md5sum

tgolsson, 2015-11-12:
__Note:__ This could also be an issue for ros/actionlib or ros/geometry. They contain the messages causing the faulty behaviour. However, I am posting it here now as this is the common denominator. 

Issue
---
Inside folder Arduino/libraries, running `rosrun rosserial_arduino makelibrary.py .` will cause some packages to fail with the error `('NoneType' object has no attribute '_md5sum')`.

The full backtrace:
```python
*** Warning, failed to generate libraries for the following packages: ***
    tf ('NoneType' object has no attribute '_md5sum')
    actionlib ('NoneType' object has no attribute '_md5sum')
Traceback (most recent call last):
  File "/home/tgo/catkin_ws/src/rosserial/rosserial_arduino/src/rosserial_arduino/make_libraries.py", line 91, in <module>
    rosserial_generate(rospack, path+"/ros_lib", ROS_TO_EMBEDDED_TYPES)
  File "/home/tgo/catkin_ws/src/rosserial/rosserial_client/src/rosserial_client/make_library.py", line 584, in rosserial_generate
    raise Exception("Failed to generate libraries for: " + str(failed))
Exception: Failed to generate libraries for: ["tf ('NoneType' object has no attribute '_md5sum')", "actionlib ('NoneType' object has no attribute '_md5sum')"]
```
Affects
---
_tf_: breaks on /opt/ros/share/tf/msg/tfMessage.msg
Version: 1.11.7-3 
_actionlib_: breaks on /opt/ros/share/actionlib/msg/TestResult.msg
Version: 1.11.3-3

I cannot verify whether other messages on these paths will break in a similar fashion, as the exception causes the whole package to stop generation.

Expected result
---
All messages generated. In case of failure, a more descriptive error and/or fallback behaviour.

Environment and reproduction
---
I am running ArchLinux (x86_64), with ROS Indigo instaled from the ArchLinux User Repositories. As rosserial-arduino is not in the AUR, it is installed from GitHub using the catkin method. My PYTHONPATH is set to Python2.7. As per the installation guide; I have also set my catkin_make to compile with the python 2.7 flags. 

I compile rosserial inside my workspace using `catkin_make`, and then source it with `source devel/setup.bash`. I then change to the Arduino/libraries folder, and attempt to generate the libraries. 

I have verified that the offending files exist on this path, and that they have the same permissions as the ones that do work.

Attempted solutions
---
+ Redownloading/compiling ROS-serial
+ Reinstalling actionlib, tf
+ Reinstalling catkin and related packages
+ Reinstalling most of ros-indigo

alexkonradi, 2015-12-01:
I had the same issue. For me, it was caused by my Arduino package depending on a package with custom messages, but not having that dependency listed in the `package.xml`. Catkin was trying to generate the rosserial headers before the Python messages files were generated, so the `roslib.message.get_message_class` call [here](https://github.com/ros-drivers/rosserial/blob/9fca92f3a4085a59bb708c528d85831858e4b0c3/rosserial_client/src/rosserial_client/make_library.py#L506) was returning None.

tgolsson, 2015-12-01:
@alexkonradi The messages generating the error (actionlib, tf) are not a dependency of my project. I have several hundred ROS-packages installed on my computer, and all messages are generated when creating the ros_lib. 

I guess your error might be related; but I know the .py-files exist. If none of the maintainers have time to look at it, I might look at it more during the holidays.

zchen99, 2016-01-22:
Any update on this? Thanks.

zchen99, 2016-01-22:
I met the same issue. 
ROS indigo:
rosserial 0.7.1
python 2.7.6

rosrun rosserial_embeddedlinux make_libraries.py .

caused:  
*** Warning, failed to generate libraries for the following packages: ***
 bhand_controller('NoneType' object has no attribute '_md5sum')

zchen99, 2016-01-22:
Ubuntu 14.04.3

davidshin172, 2016-09-09:
I had issues with this in my build, so I'd like to share my investigation and workaround.
Investigation
----------------
The issue seems to boil down to the fact that the messages are generated, but in a development build, they don't get installed into ROS' PYTHONPATH. From this, one would expect that after a failed invocation of `catkin_make`, sourcing `devel/setup.bash` and then invoking `catkin_make` again would cause the second build to succeed.
Above seems to be the case for trusty/indigo, but not for xenial/kinetic. My investigation seems to indicate that when `make_libraries.py` is eventually invoked, PYTHONPATH does not include the path sourced by `devel/setup.bash`.
Workaround
----------------
I ended up making the firmware target not part of the ALL build by removing ALL from:
```
rosserial_add_client_target(firmware <firmware project name> ALL)
```
After invoking `catkin_make` (Which no longer invokes `make_libraries.py`), source `devel/setup.bash`.
Then, invoke:
```
rosrun rosserial_arduino make_libraries.py /path/to/workspace/build/<catkin project name>
```
This generates the messages, with PYTHONPATH pointing to the ROS installation path AND the development build path.
Then, invoke the following commands to build the firmware:
```
catkin_make <catkin project name>_firmware_<firmware_project_name>
```
I've automated this to a script, and it seems to be working quite well; I'm not too familiar with the build infrastructure, so this is likely a horrible hack, but it works.

nikonikolov, 2016-10-12:
I had the same issue. For me the problem was that the core ROS packages built properly, but were split in  the folders `~/ros_catkin_ws/install_isolated/lib/python2.7/site-packages` and `~/ros_catkin_ws/install_isolated/lib64/python2.7/site-packages`. Thus e.g. there would be a module named `actionlib` in both  `~/ros_catkin_ws/install_isolated/lib/python2.7/site-packages` and `~/ros_catkin_ws/install_isolated/lib64/python2.7/site-packages`, but the module will have different submodules in each directory. As both of the directories are on the `$PYTHONPATH`, python finds only one of the `actionlib` modules and the other one is ignored. This is not a python issue, but an issue with how ROS built its modules.

The solution is to put all modules in the same directory so python does not get confused. This is the way it is supposed to be. Here is a possible way to do that:

```
#!/bin/bash
SOURCE_PATH=~/ros_catkin_ws/install_isolated/lib/python2.7/site-packages
DEST_PATH=~/ros_catkin_ws/install_isolated/lib64/python2.7/site-packages

for dir in $SOURCE_PATH/*/ ; do

    dir_base="$(basename $dir)"

    if [ -d $DEST_PATH/$dir_base ]; then
    	mv $dir/* $DEST_PATH/$dir_base
    	rm -rf $dir
    else
    	mv $dir $DEST_PATH/
    fi
done
```

The above script will put everything under the proper `lib64` subdirectory. 

I also submitted an issue for catkin https://github.com/ros-infrastructure/catkin_pkg/issues/153

maxb91, 2016-10-21:
In case someone else experiences this issue: In my case I had some custom messages that were **not** added in my CMakeLists.txt (just old msg-files I didn't use anymore). Deleting these files solved the issue.

PR #257 (2016-11-29): Fixing message has no attribute _md5sum
If a message package has issues with it being unable to be imported, message generation will fail with: `message has no attribute _md5sum`. Now skip messages that are unable to be imported with a error message to the build log.

This fixes #199.
commit 6c0864df (2016-11-29): Fixing message has no attribute _md5sum (#257)

If a message package has issues with it being unable to be
imported message generation will fail with: "message has no attribute
_md5sum". Now skip messages that are unable to be imported.

This fixes #199.
closed 2016-11-29 by mikepurvis