ros-drivers/rosserial #205: Unable to install ros_lib into Arduino IDE: rosserial_client (missing dependency)

chrisspen, 2015-12-14:
I'm trying to follow the [tutorial](http://wiki.ros.org/rosserial_arduino/Tutorials/Arduino%20IDE%20Setup#A.28RECOMMENDED.29_Installing_Binaries_on_the_ROS_workstation) for installing ROS into the Arduino IDE.

I installed the binary packages with:

     sudo apt-get install ros-indigo-rosserial-arduino ros-indigo-rosserial ros-indigo-rosserial-client

but when I run the command to copy the files into the Arduino libraries folders with:

     rosrun rosserial_arduino make_libraries.py .

it partially fails with the error:

    Exporting to .
    [rosserial_client]: Unable to find dependency: angles
    ROS path [0]=/opt/ros/indigo/share/ros
    ROS path [1]=/opt/ros/indigo/share
    ROS path [2]=/opt/ros/indigo/stacks. Messages cannot be built.

    Exporting rosserial_msgs

      Messages:
        Log,TopicInfo,

      Services:
        RequestParam,RequestServiceInfo,RequestMessageInfo,

    Exporting std_srvs

      Services:
        Empty,Trigger,

    Exporting std_msgs

      Messages:
        UInt8MultiArray,UInt16MultiArray,MultiArrayDimension,Byte,Int16,UInt64MultiArray,Int8,Int64MultiArray,Char,Duration,Float64MultiArray,UInt32MultiArray,Int32MultiArray,Header,Int16MultiArray,String,UInt8,ColorRGBA,ByteMultiArray,MultiArrayLayout,Float32MultiArray,UInt64,UInt32,Time,Int32,UInt16,Float32,Empty,Float64,Int64,Bool,Int8MultiArray,

    Exporting bond

      Messages:
        Status,Constants,

    Exporting diagnostic_msgs

      Messages:
        KeyValue,DiagnosticStatus,DiagnosticArray,

      Services:
        SelfTest,

    Exporting rosgraph_msgs

      Messages:
        Clock,Log,TopicStatistics,

    Exporting geometry_msgs

      Messages:
        PoseWithCovarianceStamped,Point32,Transform,Wrench,InertiaStamped,TwistStamped,PoseArray,Pose,Point,AccelWithCovarianceStamped,PointStamped,Quaternion,Pose2D,PoseStamped,QuaternionStamped,TwistWithCovariance,Vector3Stamped,WrenchStamped,PolygonStamped,PoseWithCovariance,Twist,Accel,TwistWithCovarianceStamped,Inertia,AccelStamped,TransformStamped,Polygon,Vector3,AccelWithCovariance,

    Exporting sensor_msgs

      Messages:
        JoyFeedback,MagneticField,ChannelFloat32,LaserEcho,RelativeHumidity,RegionOfInterest,FluidPressure,PointField,MultiEchoLaserScan,Joy,Temperature,LaserScan,NavSatFix,TimeReference,PointCloud2,JointState,CameraInfo,Image,Illuminance,JoyFeedbackArray,Range,NavSatStatus,CompressedImage,Imu,PointCloud,MultiDOFJointState,

      Services:
        SetCameraInfo,

    Exporting visualization_msgs

      Messages:
        Marker,InteractiveMarkerPose,InteractiveMarkerControl,InteractiveMarkerInit,ImageMarker,InteractiveMarker,MenuEntry,InteractiveMarkerUpdate,InteractiveMarkerFeedback,MarkerArray,

    Exporting shape_msgs

      Messages:
        MeshTriangle,Mesh,SolidPrimitive,Plane,

    Exporting trajectory_msgs

      Messages:
        JointTrajectory,MultiDOFJointTrajectory,MultiDOFJointTrajectoryPoint,JointTrajectoryPoint,

    Exporting actionlib_msgs

      Messages:
        GoalStatus,GoalID,GoalStatusArray,

    Exporting nav_msgs

      Messages:
        Path,GetMapAction,OccupancyGrid,GetMapActionFeedback,MapMetaData,Odometry,GetMapResult,GetMapActionResult,GetMapGoal,GetMapActionGoal,GetMapFeedback,GridCells,

      Services:
        SetMap,GetPlan,GetMap,

    Exporting tf2_msgs

      Messages:
        TF2Error,LookupTransformResult,TFMessage,LookupTransformActionFeedback,LookupTransformActionGoal,LookupTransformAction,LookupTransformActionResult,LookupTransformGoal,LookupTransformFeedback,

      Services:
        FrameGraph,

    Exporting stereo_msgs

      Messages:
        DisparityImage,

    Exporting roscpp

      Messages:
        Logger,

      Services:
        GetLoggers,Empty,SetLoggerLevel,

    Exporting rosserial_arduino

      Messages:
        Adc,

      Services:
        Test,

    Exporting nodelet

      Services:
        NodeletList,NodeletUnload,NodeletLoad,

    Exporting topic_tools

      Services:
        MuxDelete,DemuxSelect,MuxSelect,DemuxList,DemuxAdd,MuxAdd,MuxList,DemuxDelete,

    Exporting dynamic_reconfigure

      Messages:
        ParamDescription,StrParameter,BoolParameter,Config,GroupState,DoubleParameter,SensorLevels,IntParameter,ConfigDescription,Group,

      Services:
        Reconfigure,

    Exporting actionlib

      Messages:
        TwoIntsActionResult,TestRequestActionResult,TestRequestAction,TestRequestFeedback,TwoIntsAction,TestRequestActionGoal,TestGoal,TwoIntsActionFeedback,TestFeedback,TestActionGoal,TestActionFeedback,TwoIntsActionGoal,TestResult,TwoIntsResult,TwoIntsGoal,TwoIntsFeedback,TestRequestResult,TestRequestGoal,TestRequestActionFeedback,TestAction,TestActionResult,

    Exporting tf

      Messages:
        tfMessage,

      Services:
        FrameGraph,

    *** Warning, failed to generate libraries for the following packages: ***
        rosserial_client (missing dependency)

How do I fix my install so rosserial_client can be used?

What package provides "angles" and is it an apt or pip package? There does exist an "angles" package in pypi, but installing this seems to have no effect.

mikepurvis, 2015-12-14:
Hmm, angles is [this package](http://wiki.ros.org/angles), but it's [not listed as a dependency of rosserial_client](https://github.com/ros-drivers/rosserial/blob/jade-devel/rosserial_client/package.xml), so I'm confused by why that would be an issue.

chrisspen, 2015-12-14:
Yes, installing that package fixed the error, so it seems that package is somehow being listed as a dependency in indigo.