ros-drivers/rosserial #206: Unable to sync with Arduino Leonardo

chrisspen, 2015-12-14:
I managed to get the [Blink](http://wiki.ros.org/rosserial_arduino/Tutorials/Blink) example to compile fine, but when I run `rosrun rosserial_python serial_node.py /dev/ttyACM0`, it just outputs a series of errors:

    [ERROR] [WallTime: 1450067688.584357] Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

My target is an Arduino Leonardo, and [this post](https://defendtheplanet.net/2014/05/22/arduino-microleonardo-and-rosserial-hello-world-possible-link-problem/) suggested that the sample code is incompatible and needs to have the line `#define USE_USBCON` added to the top. However, adding this line did not fix the problem for me.

I also confirmed there are no other ROS installations, so the Arduino and my host PC are both using Indigo.

AlexisTM, 2015-12-22:
Dear, I got the exact same problem, I re-cloned the `indigo` version and I got still the same problem. 

```
git clone --branch indigo-devel https://github.com/ros-drivers/rosserial.git
```

Now, I use the C++ implementation and it seems to work

```
rosrun rosserial_server serial_node /dev/ttyACM0
``` 

Maybe the Python implementation is well implemented for indigo ?

Shawnnn, 2017-01-09:
Possible problem:
Have u ever change the baud rate of serial_node.py? It is set to default value, 57600.
If so, it might be the unidentical baud rate that cause the problem.
you can change your command to run rosserial_python to make sure the baud rate is identical to that of arduino code
 `rosrun rosserial_python serial_node.py /dev/ttyACM0 _baud:=57600` 
Correspondingly, in arduino code, add below code to setup() function:
`Serial.begin(57600)`

chrisspen, 2017-06-26:
No, I'm setting the baudrate of 57600 on both the Arduino and the ROS node.

It starts fine, but even though I've gutted the Arduino code to do nothing more than register some subscribers, which are never called, and run spinOnce(), it almost immediately loses connection:

```
[INFO] [1498440030.348370]: ROS Serial Python Node
[INFO] [1498440030.352508]: Connecting to /dev/ttyACM0 at 57600 baud
[INFO] [1498440032.681094]: Note: subscribe buffer size is 512 bytes
[INFO] [1498440032.681402]: Setup subscriber on pan/set [std_msgs/Int16]
[INFO] [1498440032.693910]: Setup subscriber on tilt/set [std_msgs/Int16]
[INFO] [1498440032.700221]: Setup subscriber on halt [std_msgs/Empty]
[ERROR] [1498440047.677618]: Lost sync with device, restarting...
[ERROR] [1498440062.789621]: Lost sync with device, restarting...
```

My Arduino code is only using 48% of memory, so I don't think it's a low-memory issue either.

I sometimes get errors like:

```
[WARN] [1498439316.019368]: Serial Port read returned short (expected 68 bytes, received 62 instead).
[WARN] [1498439316.020448]: Serial Port read failure: 
[INFO] [1498439316.021580]: Packet Failed :  Failed to read msg data
[INFO] [1498439316.022394]: msg len is 72

```

so I think it's an issue with a very small buffer size on the Leonardo. Unfortunately, this seems to be hardcoded in rosserial.

I've implemented my own slimmer version of rosserial, and it works fine. I just wish I didn't have to re-invent the wheel. Could the authors add a doc page listing which Arduinos are compatible with this package, and that the Leonardo and Uno do not work with Rosserial?

AlexisTM, 2022-12-05:
This is 2017 ;)

closed 2022-12-05 by mikepurvis