ros-drivers/rosserial #213: Problem in launching rosserial_python

gokoreas, 2016-02-08:
The rosrun to run the rosserial_python works after I have installed rosserial

However, I cannot launch rosserial_python using roslaunch

```
<launch>

<node pkg="rosseiral" type="serial_node.py" name="serial_node">
	<param name = "port" value ="/dev/ttyACM0" />
	<param name = "baud" value = "115200" />
</node>

</launch>

```
How can I resolve the problem? Is the problem from the ROS_PACKAGE_PATH?

[image]

tonybaltovski, 2016-02-08:
Typo, should be rosserial not rosseiral.

Jaeyoung-Lim, 2016-02-21:
@tonybaltovski  

Thanks for the help,

don't understand why I couldn't have found it...

closed 2016-02-21 by Jaeyoung-Lim