ros-drivers/rosserial #214: Connection Issues: rosserial and Arduino DUE

codysmithd, 2016-02-25:
I'm using an Arduino DUE and rosserial (indigo), and am having connection issues with rosserial. First, even the HelloWorld example does not work until `#define USE_USBCON` is added to the Arduio code.

When the node is started:
`rosrun rosserial_python serial_node.py /dev/ttyACM0`

I get the following error:
`Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino`

I have used both apt-get install, and built from source, and have verified in both cases that the versions are all the same and I still get the error.

The connection works for a while however. Then after some time it starts having errors, these include sync issues as well as info about the wrong checksum:

`Mismatched protocol version in packet: lost sync or rosserial_python is from different ros release than the rosserial client`

`wrong checksum for msg length, length -32255`

katodo, 2016-08-15:
Hi, I've the same problem, I need to move a SW based on [Arduino Oscilloscope](http://wiki.ros.org/rosserial_arduino/Tutorials/Arduino%20Oscilloscope) example from an Arduino Mega to an Arduino DUE because I need 12bit adc.
In my code I've used "_#define USE_USBCON_" declaration.
When I power on board I see 2 led on; Green=Power and Yellow=Classic Pin 13 led.

I start node with _rosrun rosserial_python serial_node.py /dev/ttyACM1_ command and I see power off pin 13 led and this messages:

`max_xxv@plutarco:~$ rosrun rosserial_python serial_node.py /dev/ttyACM1
[INFO] [WallTime: 1471245142.265776] ROS Serial Python Node
[INFO] [WallTime: 1471245142.290537] Connecting to /dev/ttyACM1 at 57600 baud
[ERROR] [WallTime: 1471245159.402578] Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
[INFO] [WallTime: 1471245159.763088] Note: publish buffer size is 512 bytes
[INFO] [WallTime: 1471245159.764457] Setup publisher on adc [rosserial_arduino/Adc]
`
Have you any suggestions?

bhavyangupta, 2016-08-15:
Hi @katodo 

The fact that `serial_node` actually sets the publish/subscribe buffer size indicates that the communication between Due and your laptop is working - at least until you call `initNode()`. The sync issues can come because of many reasons, here are some things to check: 

0. The rosserial version used to generate ros_lib for the microcontroller should be the same as the version that you are using for the serial node - check the version number in the package.xml for the rosserial package that generated the ros_lib and the one you are using to communicate. 
0. Are there any "hanging" function calls in your code? If the microcontroller hangs, the heartbeat between the laptop and the microncontroller is absent and serial_node assumes that there is some issue with the connection.
0. Are you calling `spinOnce()`?
0. Are you sending large arrays - i.e. greater than 256 elements in length?