ros-drivers/rosserial #220: [rosrun] Couldn't find executable

mamdouh1, 2016-04-26:
i got this error while trying to generate custom message for Arduino 
`$rosrun rosserial_client make_library.py Arduino/libraries/ros_lib/ catkin_ws/src/beginner_tutorials/msg/MAAS.msg`

`[rosrun] Couldn't find executable named make_library.py below /opt/ros/indigo/share/rosserial_client`

vmatos, 2016-06-21:
Could you try: `rosrun rosserial_arduino make_library.py` ?

emilyfy, 2017-10-25:
Hello, I have the same issue for rosserial_python. When running
`rosrun rosserial_python serial_node.py`
I get an error message
`[rosrun] Couldn't find executable named serial_node.py below /opt/ros/indigo/share/rosserial_python`

I have made sure to source both
`source /opt/ros/indigo/setup.bash`
`source ~/catkin_ws/devel/setup.bash`
in my ~/.bashrc

Really couldn't figure out why this happens cause the serial_node.py is definitely inside my /opt/ros/indigo/lib folder. The only way I could finally run the node is if I clone the rosserial_python package into my catkin_ws/src folder so that it can be found inside my workspace. Not an elegant solution but it works, I guess. You can try doing so too if you just want to get it to work, but I've also noticed that the executable inside rosserial_client is actually called make_libraries. You could maybe try
`rosrun rosserial_client make_libraries`
and see if it works?
And yeah as vmatos said, to generate custom messages for Arduino you can simply run
`rosrun rosserial_arduino make_libraries.py`
That worked for me.

Hope that helps!

mikepurvis, 2017-10-25:
Hey @emilyfy, thanks for the report here. Can you clarify if it works when you _only_ `source /opt/ros/indigo/setup.bash`? I'm wondering if there's an issue with `rosserial_python` being in your overlay as un-built source, which can alias binaries in underlay workspaces.

Also, note that you should normally only have to source the "top" workspace in the case of an overlay. The underlay workspace is implicitly included in your environment.

emilyfy, 2017-10-26:
Hi @mikepurvis, thank you I finally got it to work! At first I tried sourcing only `source /opt/ros/indigo/setup.bash` and it didn't work either. When you said I should only have to source the top workspace, does it mean /opt/ros/indigo/setup.bash is enough?

But in any case, I got it to work in the end even with both setup.bash sourced. The thing is, it used to work for me all along until one day when I cloned a repo containing whole rosserial package into my catkin_ws. I think it hasn't been working since then even though I've deleted the whole package from my catkin_ws and also their corresponding build and install directories. But today I tried running catkin_make clean and deleting all files and directories containing 'rosserial' from my catkin_ws. It works perfectly now. Thanks again!

mikepurvis, 2017-10-26:
Cool, good to know.

closed 2017-10-26 by mikepurvis

surabhi96, 2017-12-14:
@emilyfy @mikepurvis 
I am getting the same problem and entered the following two commands
`rosrun rosserial_client make_libraries`
`rosrun rosserial_arduino make_libraries.py`
and it says 
`This generic make_libraries generates the non-platform specific part of a
rosserial client library, including messages. It assumes the target platform
has native float64 support, and is therefore unsuitable for use with AVR.
To generate messages for AVR, use the make_libraries script in
rosserial_arduino.

This script also does not provide the ros.h file or other hardware
abstraction. This will need to be supplied by the user.

rosrun rosserial_client make_libraries <output_path>
`
`make_libraries.py generates the Arduino rosserial library files.  It 
requires the location of your Arduino sketchbook/libraries folder.

rosrun rosserial_arduino make_libraries.py <output_path>`

So I enter the following command
`rosrun rosserial_arduino make_libraries.py ~/sketchbook/libraries/ adc_voltage
`
to create the required headers for Arduino and add them to the "ros_lib" folder under sketchbook/libraries folder.

but I get the following 
`Traceback (most recent call last):
  File "/opt/ros/indigo/share/rosserial_arduino/make_libraries.py", line 87, in <module>
    shutil.copytree(rosserial_arduino_dir+"/src/ros_lib", path+"/ros_lib")
  File "/usr/lib/python2.7/shutil.py", line 177, in copytree
    os.makedirs(dst)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 17] File exists: '/home/surabhi/sketchbook/libraries/ros_lib'
`
I am not able to generate the required header in any case. Please help!

Thanks

romainreignier, 2017-12-14:
The issue is clear enough:

> [Errno 17] File exists: '/home/surabhi/sketchbook/libraries/ros_lib'

You have to remove the existing `ros_lib` folder first.

surabhi96, 2017-12-20:
This worked! Thanks a lot