ros-drivers/rosserial #223: ros kinetic

pavoldroba, 2016-05-27:
Hi,

is there a plan to release this package for ROS Kinetic?

Thanks,
Pavol

zaidpirwani, 2016-05-30:
Yeah, I would like to know as well, I just started with ROS, first installed kinetic, but then installed Jade cause I need to use rosserial.

Planning to make a robot with Arduino and Raspberry Pi and then have the bot communicate with other SAME bots over WiFi and BLE

mikepurvis, 2016-05-31:
I need to do some smoke testing of the merged changes but then, yes, there will be a new release for Jade and Kinetic.

robin-read, 2016-06-16:
Do you have an estimated release date for this at all?

mikepurvis, 2016-06-16:
If you would be willing to build the current head on Kinetic and verify basic functionality (eg, Arduino publish/subscribe) with rosserial_server and rosserial_python, that would be very helpful.

robin-read, 2016-06-16:
About to head off travelling for a week, so I'll have a quick look now, but realistically you'll have an answer in about a week and a half.

vmatos, 2016-06-21:
I've just installed a fresh Xubuntu 16.04 + ROS kinetic.
Just built rosserial fine with no errors. Later today I'll update my arm toolchain for the new gcc arm PPA, and then I'll test rosserial_tivac + update tutorials.

Anything I should be looking out for?

vmatos, 2016-06-23:
**1 -** I think I came across with something on `rosserial_client/cmake/rosserial_client-extras.cmake` that needs changing for kinetic: https://github.com/robosavvy/rosserial/issues/1

**2 -** The other issue I'm having, is a different behavior when building `rosserial_client` packages when messages are defined on the parent package. For instance my https://github.com/robosavvy/rosserial_tivac_tutorials package.

On Jade, I just make `catkin_make`, custom messages are generated first, are copied to `devel`, then `rosserial_client` uses them when generating `ros_lib` folder. Even if I have not sourced `devel/setup.bash`.

On Kinetic, on the first `catkin_make`,  custom messages are generated and copied to `devel`, but during `rosserial_tivac_tutorials_ros_lib`, on `make_library.MakeLibrary()`, the messages are not found by `roslib`.
```
md5sum = roslib.message.get_message_class(package+'/'+f[0:-4])._md5sum
```
returns None.

robin-read, 2016-06-28:
So what I've just tested is cloning the latest rosserial repos and adding that to my Kinetic catkin workspace. `catkin_make` built without issue. I've connected my computer to an arduino (controlling the wheels of my robot) and have been able to publish Twist messages to the arduino, and receive back multiarray messages from the arduino with encoder tick counts.

Haven't tried making any custom messages for this, and everything has worked out of the box thus far. But I have only been using things that came in the box... :)

khancyr, 2016-06-28:
Hi,
I use it on kinetic for some time. On arduino side, it seems ok. I didn't catch new problem.
Catkin build work fine too
But gcc6.1 trigger many errors...

vmatos, 2016-06-30:
Has anyone tried to use `rosserial` capabilities using `rosserial_client`?
e.g. just like the tutorial for Arduino using cmake: http://wiki.ros.org/rosserial_arduino/Tutorials/CMake

I've just published a minimum example which reproduces my error. Can you try it?
https://github.com/vmatos/rosserial_ard_ws/

mikepurvis, 2016-08-05:
Rosserial has been released to Kinetic.

closed 2016-08-05 by mikepurvis