ros-drivers/rosserial #229: rosserial_server's serial_node doesn't recognize get param

vmatos, 2016-07-07:
Couldn't get parameter from master using rosserial_server.
Works fine with rosserial_python.

Related: http://answers.ros.org/question/238879/cannot-read-rosserial-getparam/?answer=238985

vmatos, 2016-07-07:
My take on the handler for getting the parameter from the master: https://github.com/robosavvy/rosserial/commit/80162f0db4c4a4ea9b32364593e8c9bc1443f8cb

It's not the most beautiful implementation. And I'm getting trouble having XmlRpc detecting parameters as arrays/lists and not strings.

vmatos, 2016-07-11:
Handler for getparam for `rosserial_server` session: https://github.com/robosavvy/rosserial/commit/7e3333ebd5e7a3996027b41cb51d6051a350d480

Fully tested with example on getparam demo for `rosserial_tivac`: https://github.com/robosavvy/rosserial_tivac_tutorials/tree/master/getparam
Launch file for the test: https://github.com/robosavvy/rosserial_tivac_tutorials/blob/master/launch/test_param.launch

agjunyent, 2016-07-11:
Will test it when I get home and update answer (I'll update answers.ros also)

martonmiklos, 2017-04-09:
Hi guys,
We have the same issue, do you have any feedback on this one?
Any plans on merging?

senceryazici, 2020-01-29:
Is this alive ? I have the same problem, works with serial_node.py, fails with socket_node

martonmiklos, 2020-01-29:
@senceryazici I have submitted @vmatos  changes in #421  but it did not get merged because I have not managed to get autotests working properly.

senceryazici, 2020-01-29:
I see, running serial_node.py with port as "tcp" seems to be a temporary solution for me. Thanks..

JackFrost67, 2020-02-04:
Same problem, I figured out that there is no callback for the ID_PARAMETER_REQUEST. 
I've test @vmatos implementation, but it doesn't works (I got thousands of unrecognized topicId for each topic I've). I'll try serial_node.py