ros-drivers/rosserial #232: Can't get laserscan data using rosserial_server socket_node

imranz30, 2016-07-13:
Hi,
I am using Visual Studio on the non-ROS computer and I subscribed to a /scan topic in the ROS computer running gazebo and a robot. But I cannot get anything on the windows terminal when I run my application. Here is the code I wrote: 

[image]

On the Ubuntu machine I installed the rosserial for jade as "sudo apt-get install ros-jade-rosserial-windows" and "sudo apt-get install ros-jade-rosserial-server". On the ROS computer I am running rosserial_server socket_node command which is listening to port 11411 and I am connecting to that port through non-ROS computer. I don't understand why it doesn't receive scan data. It seems like that the callback function laser_scan_callback is never invoked.  I also tried to publish data on a topic and that worked. I published on the topic /cmd_vel and the robot in gazebo moved. But I don't know why I don't receive the data from /scan topic.

What can I do?

imranz30, 2016-07-15:
I created my own laser_scan_publisher node in ros and ran it to publish data on /scan topic. And now when I connect through rosserial_windows I am able to get that data in the terminal. I think then the problem is not with rosserial_windows. What could be the problem then? Is there something wrong with the gazebo plugin that publishes data on the /scan topic, should I change my Gazebo version or something?

imranz30, 2016-07-15:
Well I changed the sampling rate of the laser scanner to 100 from 720 and now I receive the data. Is it a performance issue or something?

yijiepan, 2016-11-13:
Hi!
I encountered the same problem as you.When I want to subscriber the topic of /turtle1/cmd_vel, but I don't receive the data.
What can I do?

imranz30, 2016-11-13:
Hi,
Try reducing the sampling rate of laser. I did that and I started receiving
data. By default it is set to 720 samples per scan which makes the data too
big for rosserial to handle. Make it 100 or something.

Hope this helps.
Imran

yijiepan, 2016-11-13:
Thank you very much!

yijiepan, 2016-11-13:
Hi，
how much information you can transfer every second, I just transfer 300 byte.

yijiepan

imranz30, 2016-11-13:
I don't know, I haven't experimented with that.

imranz30, 2016-12-15:
Hi 
Do you know how to close the socket connection from the Visual Studio application. I am creating a connection with the socket, receiving some data and no I want to close the connection. Any ideas how to do it?

Thanks,