ros-drivers/rosserial #235: rosserial will not build

jwcrawley, 2016-08-01:
Greetings,

I'm using Ubuntu 15.10, with Kinetic Kame, trying to install rosserial. Everything seems to work, and then when I go to invoke the python script to open the serial port, it fails.

Ive included logs here: http://pastebin.com/gy9Jfymi

The fail line is this

     josh@surface1:~/catkin_ws$ rosrun rosserial_python serial_node.py /dev/ttyACM0
     [rosrun] Couldn't find executable named serial_node.py below /home/josh/catkin_ws/install/share/rosserial_python

Sincerely,
Josh

jwcrawley, 2016-08-01:
I have figured out what was going on, with assistance from Freenode #ros chatroom, user Defiant.

I created the catkin_ws/src directories as stated by catkin help pages. 
I set up my .bashrc to include appropriate directories.

I then, in catkin_ws/src, git clone'd the repo.
I go back to catkin_ws
I then chown -R root:[your user] /opt/ros
I then chmod -R 0775 /opt/ros
I then catkin_make -DCMAKE_INSTALL_PREFIX=/opt/ros/kinetic install

It works by invoking "rosrun rosserial_python serial_node.py [serial port]"

closed 2016-08-01 by jwcrawley

Amr-Sayed-Shaheen, 2020-12-25:
Try to 
source /opt/ros/<distro>/setup.bash
EX:
source /opt/ros/kinetic/setup.bash
;)