ros-drivers/rosserial #239: [rosserial_client] Can't find custom message,  inside same package

vmatos, 2016-08-10:
I'm using the kinetic binary distribution, on Xubuntu 16.04.
When building the package, rosserial_client can't find the message definitions to generate the rosserial msg headers.

I'm using a minimal example from the Arduino tutorial: https://github.com/vmatos/rosserial_ard_ws/

I get the error:

```
Exporting helloworld

  Messages:
    Buttons,
[helloworld]: Unable to build messages: 'NoneType' object has no attribute '_md5sum'
```

vmatos, 2016-08-29:
Just upgraded ROS Jade packages on 14.04, and now this problem also happens on Jade.

Furthermore, it also happens even if messages are generated on other packages, prior to running `ros_lib` target, on a single `catkin_make` command.

vmatos, 2016-09-01:
I'm really confused what is causing these issues, a few weeks ago it worked fine.

I currently have two computers, 14.04 + Jade. One has ros-jade-catkin 0.6.18, and the other ros-jade-catkin 0.6.16 .

On the computer with catkin 0.6.16, both https://github.com/vmatos/rosserial_ard_ws/ and https://github.com/vmatos/rosserial_ard_ws2/ build fine.

On the computer with catkin 0.6.18, none of the two builds <pkg>_ros_lib successfully. Can't find the custom messages.

Both computers have cmake 2.8.12.2 and catkin_tools 0.4.2 and python 2.7.6.

Both computers build https://github.com/vmatos/rosserial_ard_ws2 using catkin_tools.

On the computer with catkin 0.6.18, I'm not able to build https://github.com/vmatos/rosserial_ard_ws/ with catkin tools, while on I don't get any errors on the computer with catkin 0.6.16.

mikepurvis, 2016-09-01:
Which version of rosserial is this? Note that 0.7.3 did change slightly how messages are generated, see: https://github.com/ros-drivers/rosserial/pull/234/files

vmatos, 2016-09-01:
```
vmatos@vmatos:~/ros/rosserial_ard_ws$ apt-cache show ros-jade-rosserial
Package: ros-jade-rosserial
Version: 0.7.3-0trusty-20160805-133218-0700
```

I can entirely reproduce the problem by checking out and compiling catkin 0.6.17, and then solve it by checking out and building 0.6.16.

mikepurvis, 2016-09-01:
https://github.com/ros/catkin/compare/0.6.16...0.6.17

vmatos, 2016-09-01:
Just narrowed it down to: https://github.com/ros/catkin/commit/b5ac581c807204b5333f9c29e4c1e22018ae0b88

I've looked at it a bit: https://github.com/vmatos/catkin/commit/24ceeef3715cc61f38190c81e0cf3197a88f0e7e#diff-d45f5a104dbd945efc89af62952ef042R175

From my understanding, because the `lib` folder of the workspace currently being built does not exist, it will not be added temporarily to `PYTHONPATH`. Therefore rosserial's `make_library` will never find it.

rosserial's `make_library` will try to get the message class through `roslib.message`, which will go through `genpy.message`, on function `_get_message_or_service_class` it won't be able to import the newly generated module, because the directory in not present in `PYTHONPATH`.

For a few months I've had this issue in Kinetic. I'm wondering why only came across the issue this week on Jade. And am I the only one?

mikepurvis, 2016-09-02:
Upon reflection, I think @tonybaltovski and @mikeodr have had on and off issues with this, too.

vmatos, 2016-11-30:
With catkin tools I have no better luck.

I even have
```
add_dependencies(${PROJECT_NAME}_ros_lib ${PROJECT_NAME}_generate_messages)
```
but my project's messages target is not built before the `ros_lib` target.
Although I may be missing something, because I don't usually use catkin tools.

mikeodr, 2016-11-30:
https://github.com/ros-drivers/rosserial/pull/257 should fix the error directly, bit it will still print as unable to compile bad messages.

I was seeing this error when something as part of the package the messages were contained within was unable to import.
If you try importing the package manually in a python interpreter with your workspace sourced may see an error otherwise:

```pyhton
import helloworld
# I suspect some traceback will happen here....
```
Fix the import problem and the message should probably generate.

vmatos, 2016-11-30:
Yes, it will print "unable to compile message". But that does not solve the issue of finding my custom messages within the same package, on a 'not sourced' workspace. And because ros_lib target can only run once, I can't source the  workspace and re-run ros_lib target.

1. With catkin_make, if I have the messages in the same package in a workspace I'm building from scratch, `devel/lib/python2.7/dist-packages` will not be added to $PYTHONPATH, and therefore the modules will not be imported. My workaround is in a comment above, by changing a catkin_make template.

2. With catkin_tools, if I have the messages in the same package in a workspace I'm building from scratch, I won't get message generation target before ros_lib target. When ros_lib target runs, there is no `devel/lib/python2.7/dist-packages`with messages, despite having `add_dependencies(${PROJECT_NAME}_ros_lib ${PROJECT_NAME}_generate_messages)`.

stevendaniluk, 2017-01-17:
I've encountered the problem of customs messages defined within the package as well, and have come to the same conclusion as @vmatos : that the workspace isn't sourced when the ros_lib target is run.

The `add_dependencies(${PROJECT_NAME}_ros_lib ${PROJECT_NAME}_generate_messages) ` call is in fact working properly and should be included. The problem is just that it can't see the messages.

So I believe in order to be compatible with messages within the same package, `rosserial_generate_ros_lib`, or more specifically the `add_custom_command` call, has to be modified to (re)source the appropriate workspace?

I've found one temporary workaround that doesn't involve modifying anything beyond the package. The first time building the package, or anytime the messages are changed, do the following from within your workspace:

- `catkin_make your_package`
- `source /devel/setup.bash`
- `catkin_make`

It's not the best, but it gets the proper workspace sourced.

Mayankm96, 2018-03-22:
Any updates on this? I am facing the same issue.

mattions, 2018-05-01:
Hello,
I'm having the same issue here on kinetic.

A quick way to actually get this working that worked for me is described below:

If you have a situation like

- `my_pkg`
- `my_pkg_msgs`

you can make it work doing:

1) Remove `my_pkg` from the package to build

	mv catkin_ws/src/my_pkg/package.xml catkin_ws/src/my_pkg/package.xml.tmp

Generated the messages (Note, the `my_pkg_msgs` package must be in the catkin workspace)

	catkin_make
	
Bring back my_pkg

	mv catkin_ws/src/my_pkg/package.xml.tmp catkin_ws/src/my_pkg/package.xml
	
Build everything

	catkin_make
	
If the messages do not change, you can build as normal (`catkin_make`) in the next iteration. If you are changing the messages definition you need to do the same as before

The add_dependencies and generate_msgs did not work for me.
The method proposed by @stevendaniluk  did not work as well

Basically it really boils down to not having the directory `catkin_ws/build/my_pkg/ros_lib/my_pkg_msgs/`

Note that it gets created properly if you exclude `my_pkg` build, so I'm leaning towards the idea that catkin does not resolve the dependencies in order, and runs ros_build before the messages have been generated properly.

If there is any trick to convince catkin to do that, we may be able to solve it.

I have

    <build_depend>my_pkg_msgs</build_depend>

in the `my_pgk/package.xml`, but it does not look like it gets respected.

maartendemunck, 2018-12-14:
The most elegant workaround I found so far is (compared to the [tutorial](http://wiki.ros.org/rosserial_arduino/Tutorials/CMake)):

- Remove the `ALL` property from the `rosserial_add_client_target` command for the firmware itself. Apparently, it isn't even needed to add the `EXCLUDE_FROM_ALL` property.
- Build the package using `catkin_make`
- Source the workspace's environment.
- Build the firmware using `catkin_make <package_name>_firmware_<firmware_target>`
- Upload the firmware using `catkin_make <package_name>_firmware_<firmware_target>-upload`

Compared to the approach suggested in the tutorial, I have to build the firmware manually, but at least `catkin_make` didn't fail and I didn't have to delete subdirectories in the `build` folder to continue.

charles-maheu, 2021-11-11:
This is still a problem on ROS Noetic for me. Removing the `ALL` property was not enough though, I have to comment out rosserial_add_client_target(...) for catkin_make and uncomment it before running the firmware build (`catkin_make <package_name>_firmware_<firmware_target>`).