ros-drivers/rosserial #244: Target for flashing is not found when using catkin_tools

skohlbr, 2016-09-06:
So I've been successfully using a pure CMake-based setup (to not depend on the crappy Arduino IDE) using catkin_make. Uploading worked by executing this script: https://github.com/skohlbr/simple_2wd_robot/blob/master/simple_2wd_control_arduino/scripts/upload.sh

When naively replacing `catkin_make` with `catkin build` the script does not work anymore and does not find the target:
```
---------------------------------------------------------------------
[build] Found '2' packages in 0.0 seconds.                                                                                                                                 
[build] Given package 'hector_mpu6050_imu_driver_firmware_mpu6050_imu-upload' is not in the workspace
```

I've seen #189, which is closed, but no real resolution to the problem is stated there.

vmatos, 2016-09-07:
I don't usually use catkin tools, but doesn't the command `catkin build` will build a workspace or package? Can you call a CMake target with `catkin build`?

vmatos, 2016-09-07:
Try it like this to run the `upload` target of your catkin package:

`catkin build --no-deps -v <base_pkg> --make-args <upload-target>`

I'm assuming your package is called `hector_mpu6050_imu_driver`, and the rosserial subproject you're trying to upload is called `mpu6050_imu`. If not, change it accordingly.

`catkin build --no-deps -v hector_mpu6050_imu_driver --make-args hector_mpu6050_imu_driver_firmware_mpu6050_imu-upload`

Let me know if it works, so I can add the instructions to the tutorial for using catkin tools.

messham, 2017-04-11:
Hi @vmatos. I had the same problem as OP and your instructions worked for me. Thanks!

vmatos, 2017-04-11:
Thanks @messham. Just added the instructions to the Tutorials: http://wiki.ros.org/rosserial_arduino/Tutorials/CMake