ros-drivers/rosserial #252: Rosserial MBED client - How to modify tx/rx pinnames and baud rate

vfdev-5, 2016-10-16:
Hello,
I would like to change pinnames and baud rate in mbed code and use this constructor of `MbedHardware` :
```
    MbedHardware(PinName tx, PinName rx, long baud = 57600)
      :iostream(tx, rx){
      baud_ = baud;
      t.start();
    }
```
However, the `NodeNandle_` interface is rather generic and I do not see how it could be possible to change these parameters at creation on node handle.
Could you please hint me on how to solve this problem ?

Cheers

vmatos, 2016-10-19:
It seems you can't . I would use defines, like they are used here: 
```
    MbedHardware()
      :iostream(USBTX, USBRX) {
        baud_ = 57600;
        t.start();
}
```

vfdev-5, 2016-10-19:
Yes, it seems so. It would be good to add a common interfacem, may be. 
There is, actually, a way to setup baud rate like that `nh_.getHardware()->setBaud(BAUD_RATE)` before `nh_.initNode()`, but for pinnames it needs some deeper hacks...

vmatos, 2016-10-19:
The thing is, rosserial is used not just for UART. Therefore it is difficult to have a common interface.

vfdev-5, 2016-10-19:
I mean something like that : 
- add a supplementary template and a constructor in the class NodeHandle_ 
```
template<class Hardware, class HardwareConf, ... >
class NodeHandle_ {
public:

NodeHandle_(const HardwareConf  & conf) : 
    hardware_(conf),
    configured_(false) 
{
...
}

...
}
```

- for each client : 
   - Define NodeHandle as, for example for MBED, :  `typedef NodeHandle_<MbedHardware, MbedHardwareConf> NodeHandle;`
   - add a constructor `Hardware(HardwareConf  & conf)`, e.g for MBED
```
struct MbedHardwareConf
{ 
    PinName tx;
    PinName rx;
    long baud;
}

MbedHardware(const MbedHardwareConf & conf) :
   iostream(conf.tx, conf.rx)
{
   baud_ = conf.baud;
   ...
}
```

Sure, that it needs small modifications for all clients and even if a specific hardware does not need a configuration.
What do you think ?

garyservin, 2017-03-08:
Hi @vfdev-5, I'm not sure if I understand what you need, but normally, for changing the pinnames and baudrate I do:

```
// Create the ROS node handle
class HaseHardware : public MbedHardware
{
public:
  HaseHardware(): MbedHardware(ROSSERIAL_TX, ROSSERIAL_RX, 57600) {}; 
};
ros::NodeHandle_<HaseHardware> nh;
```

vfdev-5, 2017-03-08:
@garyservin, Thanks for the answer. Right, we can derive from `MbedHardware`.

closed 2017-03-08 by vfdev-5

samchoy88, 2018-05-08:
It Works!! Below is my code for ROS changing serial port pin

```
// Create the ROS node handle
class mySTM32 : public MbedHardware
{
public:
  mySTM32(): MbedHardware(SERIAL_TX_PIN, SERIAL_RX_PIN, 57600) {}; 
};
ros::NodeHandle_<mySTM32> nh;

ros::NodeHandle  nh;

std_msgs::String str_msg;
ros::Publisher chatter("chatter", &str_msg);

char hello[13] = "hello world!";

DigitalOut led = LED_L2;

int main() {
		nh.initNode();
    nh.advertise(chatter);
    

    while (1) {
        led = !led;
        str_msg.data = hello;
        chatter.publish( &str_msg );
        nh.spinOnce();
        wait_ms(1000);
    }
}
```