ros-drivers/rosserial #258: Serial Connection Refused

SZanlongo, 2016-11-29:
Hey there, thanks for the effort put into this project.

I'm having some trouble reading the serial output from an Arduino.
The console output is below:

    $ rosrun rosserial_python serial_node.py _port:=/dev/ttyACM0 _baud:=115200
```
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/rosserial_python/serial_node.py", line 46, in <module>
    rospy.init_node("serial_node")
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 323, in init_node
    _init_node_params(argv, name)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 186, in _init_node_params
    set_param(rosgraph.names.PRIV_NAME + param_name, param_value)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 504, in set_param
    _param_server[param_name] = param_value #MasterProxy does all the magic for us
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/msproxy.py", line 148, in __setitem__
    self.target.setParam(rospy.names.get_caller_id(), rospy.names.resolve_name(key), val)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1301, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1448, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 1013, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 864, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 826, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 807, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err
socket.error: [Errno 111] Connection refused
```

I thought it could be a permissions issue, but I've checked the dialout and chmod for the port and was able to read it through the Arduino IDE serial monitor.

Any suggestions for what might be wrong?

Shawnnn, 2017-01-12:
It was probably because you forgot to rosrun roscore. Because I encountered the same problem as you due to this

SZanlongo, 2017-01-12:
Yeah, it wound up being that *roscore* wasn't running.
Thanks for following up.

closed 2017-01-12 by SZanlongo

myeabhishek, 2018-05-22:
Hello, I am facing the same issue, however, I want the serial_node to wait for roscore to start up. How can we accomplish that?