ros-drivers/rosserial #259: Teensy 3.5 and 3.6 not supported, error when compiling HelloWorld example

markwomack, 2016-11-30:
Teensy now has two new boards, 3.5 and 3.6.  When compiling the HelloWorld ros_lib example in the Arduino IDE, an error is reported (see below).

I was able to fix this by modifying [line 44 of ArduinoHardware.h](https://github.com/ros-drivers/rosserial/blob/jade-devel/rosserial_arduino/src/ros_lib/ArduinoHardware.h#L44) to be:

`#if defined(__MK20DX128__) || defined(__MK20DX256__) || defined(__MK64FX512__) || defined(__MK66FX1M0__)`

The defines of `__MK64FX512__` and `__MK66FX1M0__` are for the 3.5 and 3.6 boards, respectively.  I have only reproduced and tested this for the Teensy 3.5.

If that change can be included in the next release, then many Teensy 3.5/3.6 users will be happy.

Error reported:
Arduino: 1.6.12 (Linux), TD: 1.31, Board: "Teensy 3.5, Serial, 120 MHz, US English"

In file included from /home/mwomack/Arduino/libraries/ros_lib/ros.h:39:0,
from /home/mwomack/Arduino/libraries/ros_lib/examples/HelloWorld/HelloWorld.pde:6:
/home/mwomack/Arduino/libraries/ros_lib/ArduinoHardware.h: In constructor 'ArduinoHardware::ArduinoHardware()':
/home/mwomack/Arduino/libraries/ros_lib/ArduinoHardware.h:76:16: error: cannot convert 'usb_serial_class*' to 'HardwareSerial*' in assignment
iostream = &Serial;
^
Error compiling for board Teensy 3.5.

vmatos, 2016-11-30:
Makes sense. I would create a pull request with that correction.

markwomack, 2016-11-30:
I will try that this weekend.  I am rather new to github, so still finding my feet.

closed 2016-12-01 by mikepurvis

mikepurvis, 2016-12-01:
Change made, thanks for the suggestion!

markwomack, 2016-12-01:
One thought that came to mind on this is that instead of having to use the define for every processor type, it might be more sustainable to have something like:

#if defined(__arm__) && defined(CORE_TEENSY)

That way it will be more future proof.  But just an idea.

griswaldbrooks, 2016-12-03:
@markwomack Should a PR for the indigo-devel branch also be submitted?
Also @mikepurvis, what's the procedure for getting this supported on kinetic?

mikepurvis, 2016-12-23:
The 0.7.x series (jade-devel branch) is already released on Kinetic: http://repositories.ros.org/status_page/ros_kinetic_default.html?q=rosserial

I've picked the commit to indigo-devel in 1772985c6918fbb45109bdcaa17a368f20e33077.

griswaldbrooks, 2016-12-26:
Cool. Thanks.

commit c9e4f64a (2016-12-01): Support Teensy 3.5, 3.6.

Closes #259
commit 1772985c (2016-12-23): Support Teensy 3.5, 3.6.

Closes #259