ros-drivers/rosserial #261: [rosserial_test] new getParam tests.

vmatos, 2016-11-30:
Just wrote some tests for verifying rosserial's `getParam`. They are here: https://github.com/robosavvy/rosserial/blob/getparam/rosserial_test/src/get_param.cpp

It tests receiving a single int, float and string, as well as arrays.

My goal is to later include parameter handling in `rosserial_server` (https://github.com/robosavvy/rosserial/blob/kinetic-devel/rosserial_server/include/rosserial_server/session.h#L489).

However my tests hang when asking for nonexistent parameters and invalid arrays. (https://github.com/robosavvy/rosserial/blob/getparam/rosserial_test/src/get_param.cpp#L181)

@mikepurvis would it be interesting to include these tests? And further expand `rosserial_server` to handle parameters?

vmatos, 2016-11-30:
Just learned I must emulate the client's time tick to get a time-out on getParam.

Does it look valid? https://github.com/robosavvy/rosserial/blob/getparam/rosserial_test/src/get_param.cpp#L20

Now tests also include asking for invalid and nonexistent parameters.

mikepurvis, 2016-11-30:
Absolutely interested.

One of the big reasons that parameter support isn't part of rosserial_server is precisely that risk of hanging— it can also happen if the message in either direction gets lost. Tests would help ensure that the appropriate timeout/retry logic is in place.