ros-drivers/rosserial #262: Rosserial fail to reconnect if the device (Arduino) is unplugged.

AlexisTM, 2016-12-05:
Dear,

ROS version : Indigo, linux 14.04
ROSSerial version : Actual master (c9e4f64a6f4a55022a85c80f093c2a7fa6a84af9)

Steps to reproduce : 
* Start the computer (Odroid XU4)
* Connects the Arduino ([code](https://github.com/AlexisTM/LIDAREnhanced/blob/master/example/SixLasersROS/SixLasersROS.ino))
* `roscore` 
* `rosrun rosserial_python message_info_service.py`
* `rosrun rosserial_python serial_node.py _port:=/dev/ttyACM0 _baud:=57600`
* **It works**
* Disconnects the Arduino (the python node crashes)
* Restart the server node : `rosrun rosserial_python serial_node.py _port:=/dev/ttyACM0 _baud:=57600`
* **It do not work, failed to connect**

vmatos, 2016-12-06:
Possibly related: https://github.com/ros-drivers/rosserial/issues/177

vmatos, 2016-12-16:
Replying to your edit, could you try these steps?

*  Start the computer (Odroid XU4)
*    Connects the Arduino (code)
*    roscore
*    rosrun rosserial_python message_info_service.py
*    rosrun rosserial_python serial_node.py _port:=/dev/ttyACM0 _baud:=57600
*    Disconnects the Arduino (the python node crashes)
*    **Restart the server node : rosrun rosserial_python serial_node.py _port:=/dev/ttyACM0 _baud:=5760**
*   **Ctrl+C**
*    **Restart the server node : rosrun rosserial_python serial_node.py _port:=/dev/ttyACM0 _baud:=57600**
*    **Result?**

AlexisTM, 2016-12-16:
Dear, 

It works by killing the node via `htop`.

AlexisTM, 2024-10-28:
Outdated.

closed 2024-10-28 by AlexisTM