ros-drivers/rosserial #266: Leak of file descriptions in /rosout

nangong, 2016-12-21:
When serial_node failed to open a serial port, `  client = SerialClient(port_name, baud) ` raise an exception, I found the serial_node is not closed correctly. 

By "lsof -p <pid of rosout>", it shows an leaked file descriptor " can't identify protocol" . How to fix this problem?

mikepurvis, 2016-12-21:
The relevant code is here: https://github.com/ros-drivers/rosserial/blob/jade-devel/rosserial_server/include/rosserial_server/serial_session.h

I would have expected the socket to be destructed when the `SerialSession` instance went out of scope, but maybe it's being held onto by something? I'd be delighted for you to dig into this.

nangong, 2016-12-21:
@mikepurvis  I use the python version and I found it manually shut down a ros node by ` rospy.signal_shutdown("Error opening serial: %s" % e) #in SerialClient.py line 351` and I test a simple code: 

`
import rospy
if __name__=="__main__":
    rospy.init_node("serial_node")
    rospy.signal_shutdown("any reason")
`
which also caused a leak of file descriptor, so I believe rospy.signal_shutdown is not an appropriate approach to "shut down" a ros node. But I did not know any approach to stop a ros node by code,any idea?

nangong, 2016-12-28:
@mikepurvis 
Finally I found a tricky solution, just sleep for a second before  calling rospy.signal_shutdown and the problem disappeared.

closed 2016-12-28 by nangong