ros-drivers/rosserial #269: Unable to sync with device | mbed | LPC1768

karanchawla, 2017-02-10:
Hi,
I'm using ROS Indigo and trying to connect to mbed LPC1768 using rosserial and USB port. I'm getting a strange behavior when I compile and run HelloWorld.cpp as sometimes it connects perfectly but other times it gives the following error:
I'm running the following command:
```
rosrun rosserial_python serial_node.py _port:=/dev/ttyACM0
```

```
Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
```
These are the following things I have tried:
- Increase wait(100) to wait(10000)
- Comment out nh.SpinOnce()
- Change USB cable 

I can't find any solutions to the aforementioned problem on the internet as most of the issues mentioned are related to Arduino. Please suggest if there's something I'm doing incorrectly or if there's a fix for this solution.

ozgurerbulan, 2017-02-22:
I often encounter the same problem. I guess this is because of mbed rosserial written for jade and kinetic.

karanchawla, 2017-02-22:
Yeah I fixed the problem by moving to Kinetic.

closed 2017-02-22 by karanchawla