ros-drivers/rosserial #280: Log functions don't interpolate parameters

chrisspen, 2017-03-27:
I'm not sure if this isn't implemented yet, or I'm using it incorrectly, but the [logging page](http://wiki.ros.org/rosserial/Overview/Logging) implies the standard logging functions are supported.

However, when I try to interpolate a parameter like:

    nh.loginfo("My message: %d", somevar);

I get the error:

    main.ino:421:92: error: no matching function for call to 'ros::NodeHandle_<ArduinoHardware>::loginfo(const char [35], int)'

How do you log messages with dynamic parameters?

vmatos, 2017-03-27:
You can find the function headers here:
https://github.com/ros-drivers/rosserial/blob/jade-devel/rosserial_client/src/ros_lib/ros/node_handle.h#L468

Just like Arduino's `print`, you can't specify the format and a variable number of arguments.

For logging the way you want, first create a string and then log it using the available functions.

mikepurvis, 2017-03-27:
I've had good luck in the past using the tinyprintf implementation from here:

http://www.sparetimelabs.com/tinyprintf/tinyprintf.php

I think I would prefer to avoid adding this code to rosserial proper. Many embedded projects will have some other logging approach (eg, though serial, JTAG, UDP, whatever), and don't want to bear the costs of including this functionality directly in rosserial.

chrisspen, 2017-03-27:
Thanks. Yeah, I've gotten by with using snprintf.

closed 2017-03-27 by chrisspen