ros-drivers/rosserial #283: MSG_TIMEOUT name conflict with ChibiOS

romainreignier, 2017-04-22:
I am using ChibiOS to program STM32 microcontrollers and it already defines `MSG_TIMEOUT` that `rosserial` also defines [here](https://github.com/ros-drivers/rosserial/blob/jade-devel/rosserial_client/src/ros_lib/ros/node_handle.h#L69).

What would you think to rename it, maybe, `ROSSERIAL_MSG_TIMEOUT` to avoid any conflict because the current name is a bit too generic and may conflict with other libraries.

mikepurvis, 2017-04-24:
There's no reason for these to be `define`s at all. I would happily accept a PR that changes them into consts inside the `ros` namespace.

romainreignier, 2017-04-24:
Ok, so I will do a PR soon. Thanks.

romainreignier, 2017-05-20:
Do you think it is better to use const variables for all the constants ([like I have done here](https://github.com/romainreignier/rosserial/commit/975bba90046830a00f8bbabaa029ce5bef2ecced)) or only for the MSG_TIMEOUT one?

Maybe the ability to do a `#ifdef` should be kept but I have some doubts...

romainreignier, 2017-05-20:
Note that I have also tried to use an enum for the state machine, but an `operator++` is needed so this solution seems less interesting ([see here](https://github.com/romainreignier/rosserial/commit/2de250529402f2aa12625adf9835111fa49761fc)).

mikepurvis, 2017-05-23:
Change looks good to me. Please put up a PR!

commit 6235ffb9 (2017-09-30): Use const in ros namespace instead of #define for constants. Fix #283 (#318)
closed 2017-09-30 by mikepurvis