ros-drivers/rosserial #284: Arduiono Due Rosserial

adamagkone, 2017-04-25:
I am working with an Arduino Due board but it can cannot to ROS.Here the message displayed Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino.I already tried the suggestions on changing the ArduinoHardware.h but it still does not work.Does anyone has another soltution?
Thanks

Chiasung, 2017-05-17:
Please refer to https://github.com/ros-drivers/rosserial/issues/113

js376282, 2017-06-29:
I'm having the same issue and looking at #113 I don't see how it resolves the it.  That issue seems to have been closed and the code in ArduinoHardware.h does now have options in it for the DUE.

With that I believe it's now necessary to add a couple extra lines to take care of the DUE (see line 51 in https://github.com/ros-drivers/rosserial/blob/jade-devel/rosserial_arduino/src/ros_lib/ArduinoHardware.h).  However, I've gone through a number of permutations of such changes and still have had no success.

Chiasung, 2017-07-01:
Please try to change "Serial1" to "Serial" in the  line 75, right after /* Leonardo support */

js376282, 2017-07-02:
Thanks that works.  Would be nice if changes to the library wasn't necessary and also prevent someone else from having to hunt down the source of the problem.  Alternatives seem to be to either include:

> #define USE_USBCON
> #include <ros.h>

in that order as it seems to only work if the define is before including the ros library. 

Alternatively, I've found changing line 73 to read:

> #if defined(USBCON) and !(defined(USE_USBCON)) and !(defined(_SAM3XA_))

Both work, however, the later seems to automatically resolve issues with the DUE without having to add any code specifically for the device anywhere else.  That's said I'm not certain if this might raise an issue with something else.

W4li8, 2022-12-22:
Adding to the previous [working answer](https://github.com/ros-drivers/rosserial/issues/284#issuecomment-312502805). On the Arduino Due, this does not work with the native USB port ACM1 but only using the programming one ACM0. I gave up trying to use the other one. Also make sure to explicitly match your baud rates ;)