ros-drivers/rosserial #290: Nodehandle.connected() falsely reporting a connection

chrisspen, 2017-06-02:
I'm using an Arduino Mega, running rosserial_arduino, to control some motors, and if my host crashes or locks up, I want the Arduino to automatically stop the motors as a safety precaution.

To do this, my Arduino sketch looks like:

    ros::NodeHandle nh;

    void setup() {
        nh.getHardware()->setBaud(115200);
        nh.initNode();
        digitalWrite(13, HIGH);
    }

    void loop() {

        if (!nh.connected()) {
            motion_controller.stop();
            digitalWrite(13, LOW);
        }

        nh.spinOnce();
        delay(1);
    }

I'm running `rosserial_python` from my host with the launch file:

    <launch>
        <group ns="arduino1">
            <node pkg="rosserial_python" type="serial_node.py" name="serial_node" output="screen">
                <param name="~port" value="/dev/ttyACM0" />
                <param name="~baud" value="115200" />
            </node>
        </group>
    </launch>

Unfortunately, this doesn't work. When I run the launch file, the LED turns on, indicating the sketch is running, and my host's debugging output shows the Arduino connects. However, if I kill the rosserial node running on the host, the LED on the Arduino stays turned on, indicating `nh.connected()` is broken and incorrectly reporting that it's still connected. Am I misunderstanding how rosserial works or is this a bug?

chrisspen, 2017-06-03:
This was a problem in my code, not rosserial. Sorry for the false report.

closed 2017-06-03 by chrisspen

mikepurvis, 2017-06-04:
This issue is that `connected` takes several seconds to time out, so it's not suitable for safety use. You need to have your own higher-frequency command topic (eg, 20 Hz), with a much shorter expiry enforced in your firmware.