ros-drivers/rosserial #299: Automatics recovery from "publish before setup:q"

Baycken, 2017-07-19:


Baycken, 2017-11-03:
Fixed in #308

closed 2017-11-03 by Baycken

Nick-Zhang1996, 2018-07-03:
Hi. I'm experiencing problems similar to the scenario described here. I'm using rosserial_arduino on 328p Arduino Nano. I'm publishing a custom message and subscribing to another one. I double  checked that I sourced the setup file in both devel/ and install/. However, I get this error when running the node.
`
[INFO] [1530587602.302689]: ROS Serial Python Node

[INFO] [1530587602.305351]: Connecting to /dev/ttyUSB0 at 57600 baud

[INFO] [1530587604.410748]: Requesting topics...

[INFO] [1530587604.437979]: Note: subscribe buffer size is 300 bytes

[INFO] [1530587604.441740]: Setup subscriber on rc_vip/CarSensors [rc_vip/CarSensors]

[INFO] [1530587604.451725]: Setup subscriber on rc_vip/CarControl [rc_vip/CarControl]

[ERROR] [1530587604.747273]: Tried to publish before configured, topic id 102

[INFO] [1530587604.747689]: Requesting topics...

[ERROR] [1530587604.760458]: Tried to publish before configured, topic id 102

[INFO] [1530587604.760843]: Requesting topics...

[ERROR] [1530587604.779036]: Tried to publish before configured, topic id 102

[INFO] [1530587604.779349]: Requesting topics...

`
To be clear, I'm subscribing to CarControl and publishing to CarSensors. Here's the section of the code that's relevent:

`ros::Subscriber<rc_vip::CarControl> subCarControl("rc_vip/CarControl", &readCarControlTopic);
rc_vip::CarSensors carSensors_msg;
ros::Publisher pubCarSensors("rc_vip/CarSensors", &carSensors_msg, 1);

void setup() {

    nh.initNode();
    nh.advertise(pubCarSensors);
    nh.subscribe(subCarControl);
    

    while (!nh.connected())
        nh.spinOnce();

}

void loop() {

    pubCarSensors.publish(&carSensors_msg);
    nh.spinOnce();

}
`
I removed some code not relevant to ROS, mainly Servos and I2C stuff. I saw on https://github.com/ros-drivers/rosserial/issues/63 that I need to do advertise() before handshake, but I'm not sure when the handshake takes place. I tried putting advertise() before nh.initNode() but that doesn't change anything.

Any ideas? @Baycken