ros-drivers/rosserial #302: rosserial_mbed fails to link when compiled with gcc4mbed

naoki-mizuno, 2017-07-25:
I have encountered a problem where the linker gives an `undefined reference` error when compiling with [gcc4mbed](https://github.com/adamgreen/gcc4mbed). The description of the issue and the steps to reproduce are stated in adamgreen/gcc4mbed#64.

Following is what @adamgreen found out:

> Deep down in the newlib Standard C Library, there is a stat.h which #includes time.h and it picked up the ROS one instead of the Standard C Library version. The really freaky thing is that this caused the normalizeSecNSec() function to be declared as a C function in the global namespace instead of a C++ name mangled function in the ros:: namespace. That's why it failed to link.

Renaming `time.h` to `ros_time.h` fixed the problem, but I wasn't sure if renaming the header file may break the existing codebase. My first guess is that it's not a big deal since the users basically only include `ros.h` and not `ros/time.h` specifically, but I wanted to discuss with the maintainers before I make a PR.

naoki-mizuno, 2017-07-25:
Thanks to @adamgreen, gcc4mbed now supports the `!` prefix, which fixes this problem without having to rename the headers. Details on how to do this is described in https://github.com/adamgreen/gcc4mbed/issues/64#issuecomment-317647767.

closed 2017-07-25 by naoki-mizuno

mikepurvis, 2017-07-25:
I looked at [these search results](https://github.com/ros-drivers/rosserial/search?q=include+time.h) and didn't see anything obvious, but if there are instances where we're including any of the rosserial headers without the `ros/` path prefix, that should be fixed.

naoki-mizuno, 2017-07-25:
As far as I have searched, all the `#include` for `ros/time.h` had the `ros/` prefix. I think it's just that due to how the include path is generated (for mbed), the compiler got confused between the standard `time.h` and rosserial's `time.h`.