ros-drivers/rosserial #303: Install from source on Ubuntu 17.04

hardwi-s, 2017-07-29:
I'm trying to install from source by following the instructions at http://wiki.ros.org/rosserial_arduino/Tutorials/Arduino%20IDE%20Setup
The **catkin_make** and **catkin_make install** steps look like they are going ok, but I don't get any ros_lib in my <ws>/install. When I run 
cd <sketchbook>/libraries
rm -rf ros_lib
rosrun rosserial_arduino make_libraries.py .
I get an error **[rospack] Error: package 'rosserial_arduino' not found**

Is there a step missing from the wiki? My ros installation is Lunar Loggerhead.

I'm an experienced linux user and programmer, but I'm very new to ROS and it's admin.

tonybaltovski, 2017-07-29:
Did you source your workspace?

hardwi-s, 2017-07-30:
Thanks, sourcing setup.bash has fixed it. For the info of anyone else struggling, the ros_lib directotry is created in install/share/rosserial_arduino/src.

After sourcing setup.bash, I could cd to sketchbook/libraries and rosrun rosserial_arduino make_libraries.py . no longer gives me an error, and ros_lib is present in sketchbook/libraries.

Thanks for the help.

closed 2017-07-30 by hardwi-s