ros-drivers/rosserial #325: spinOnce timeout at 5ms

1r0b1n0, 2017-10-03:
Hi,
While browsing the code I stumbled on this line that seems like a bug :

https://github.com/ros-drivers/rosserial/blob/2f18dfa843f0d256a6674943a93bbe90061b8173/rosserial_client/src/ros_lib/ros/node_handle.h#L215

This means that the communication could be reset if we spent more than 5ms processing previous data (callbacks included).
Shouldn't it be something like 5s :
`else if( hardware_.time() - c_time > (SYNC_SECONDS*1000)){` ?

mikepurvis, 2017-10-03:
Indeed, this is a units comparison failure! Thanks for the catch— do you mind submitting a PR to correct the issue?

FYI @mikeodr

closed 2017-10-04 by 1r0b1n0

supercrazysam, 2023-09-15:
@mikepurvis may I ask what would be a reasonable value to set the spin_timeout?