ros-drivers/rosserial #330: Not compatible with Timer library

yodahuang, 2017-11-01:
`rosserial` is not capatible with [`Timer`](https://playground.arduino.cc/Code/Timer) library.

With multiple publisher existing, the behavior of Timer library will be unpredictable. It can either never run callback or run callback at random times.

An example will be [the code of my robot using Timer](https://github.com/RealCabot/arduino_node/tree/fcf276f0a43c271f9d810d859d7244c1d585c150), which will simply freeze itself, with heartbeat not working. After [removing Timer and using polling](https://github.com/RealCabot/arduino_node/commit/d22def45828261d496a5866d7b17750645b9d658), everything works fine.

Baycken, 2017-11-03:
```
Sketch uses 21,658 bytes (75%) of program storage space. Maximum is 28,672 bytes.
Global variables use 2,536 bytes (99%) of dynamic memory, leaving 24 bytes for local variables. Maximum is 2,560 bytes.
```
I compiled your code on an Arduino Uno and you are really low on memory, I suspect the issue you are encountering has little to do with compatibility with `Timer` library, but rather the extra memory used by `Timer` library caused stack overflow and broke your code.

You can try to reduce the buffer size used by `rosserial` and that should solve your problem.

yodahuang, 2017-11-10:
You're exactly right. The problem is cause by SRAM.

closed 2017-11-10 by yodahuang