ros-drivers/rosserial #331: Discussion for node_handle input buffer and output buffer usage

Baycken, 2017-11-03:
### 1. Overriding Buffer size
For Arduino with little SRAM, the memory usage for the input buffer and the output buffer could make of the difference of a system working flawlessly vs a system not working at all.

Multiple times in the past, I had to change the buffer size from the default one to a lower value, otherwise something like creating a string buffer for logging would cause stack to overflow into heap and hang my Arduino.
 
I think it will be useful to have a clear and documented way to override the buffer sizes defined in the `ros.h` and I see two options:
1. We add some overriding macro that when set, they `#undef` and re`#define` default buffer size.
2. We add some documentations explaining on how to have a local copy of `ros.h` that override the version in `rosserial_arduino`.

I could work on option 1 as that is a rather easy task to do, but I personally think option 2 is better, it is just that I don't really know how to modify the ROS wiki.

### 2. Default buffer size
Right now, the buffer sizes allocated for each MCU type seem to be a bit arbitrary.

Would it be a good idea to use minimal amount of memory, when the MCU in used for `rosserial_arduino` is not defined in `ros.h`, instead of current inheriting from `rosserial_client`? Like using 512 bytes for input and 512 bytes for output is a huge overhead in term of memory usage  for an MCU like atmega32u4 (Arduino Leonardo, 40% of the total 2.5K SRAM), where as the code could probably run fine with only 150 bytes input and 150 bytes output and save ~700 bytes of SRAM for user. 

Heck, would it be a good idea to use the bare minimal (the value used for atmega8 and 16) as default value,  and explain what will be the symptom when the sizes are to small and how to increase them as needed?

mikepurvis, 2017-11-03:
For all of Clearpath's commercial firmwares that use rosserial, we obviously have our own `ros.h`, in part because we're often using an in-house port of the client that is tied enough to the rest of the system that it would be difficult to open-source in a helpful way.

But I agree with that this option is not well communicated— it's assumed (correctly, I think) that for an elementary user, it's much preferred to have things "just work" and in particular, just work for the two entry level use cases of compiling the client in the Arduino IDE and using CMake in a catkin package.

I would be delighted for you to add content to the wiki which helps clarify this advanced-level feature, perhaps as a new tutorial? http://wiki.ros.org/rosserial/Tutorials

Baycken, 2017-11-03:
@mikepurvis I just submitted a request to be whitelisted for ROS wiki, once that is approved, I'll be working on document this feature in new tutorial.

How about second point?

Right now it is possible to know a message is too big for output buffer, and something could be implemented for the input buffer as well in [here](https://github.com/ros-drivers/rosserial/blob/jade-devel/rosserial_client/src/ros_lib/ros/node_handle.h#L234).

Variables can be add to keep track of each subscriber and publisher setup attempt, and a warning messages can be sent immediately after node configuration if there is more setup attempts than slots for subscribers and publishers, somewhere [here](https://github.com/ros-drivers/rosserial/blob/jade-devel/rosserial_client/src/ros_lib/ros/node_handle.h#L390).

If we implement those changes, then users will have a full set of error message when configurations when default `node_handle` is not sufficient to support their needs.

With the `ros.h` advanced configurations documented properly, would you think it will be a good idea to use a minimal default value (`6,6,150,150`) for all MCU, instead of current per MCU basis?
Would you think those values are enough for an average user?

The incentive for me to implement this is when MCU ran out of SRAM, it hangs with no additional informations, but when the setup is not enough, we could warn the user about it, which from a debugging point view, a much easier thing to deal with.

Baycken, 2017-11-06:
@mikepurvis I have added/updated few pages. 
Notably: 
* Moved the limitations and the description of the protocol from the dedicated overview page to sections of the main page.
* Added a tutorial on [Logging](http://wiki.ros.org/rosserial_arduino/Tutorials/Logging)
* Added a tutorial on [NodeHandle and ArduinoHardware](http://wiki.ros.org/rosserial_arduino/Tutorials/NodeHandle%20and%20ArduinoHardware).
* Touched a bit on memory usage in the [HelloWorld tutorial](http://wiki.ros.org/rosserial_arduino/Tutorials/Hello%20World)

mikepurvis, 2017-11-06:
Had a quick read-though— new content looks great, thanks!

closed 2018-01-10 by Baycken