ros-drivers/rosserial #337: Trouble receiving messages with length>16bit @ server

julled, 2017-11-16:
Hi,
i am running on the jade:devel branch and i got problems to receive occupancy gridmap messages from a rosserial_windows machine with a gridcell count bigger than 16bit. 

By using the code submitted with CR #222 i successfully created the correct headers for 32bit length messages. Additionally i increased the size of the INPUT and OUTPUT Buffers in node_handle.h . 

On the windows pc i create a occupancy grid with 100000 cells and send it via rosserial_windows. But as this results in a message with a length bigger than 16bit, it is not conform with the maximal length of a message defined by the rosserial protocol and gets dropped somewhere in rosserial_server.

Did i miss something to configure?? How is it possible to send messages longer than 16bit, as this is defined by the protocol? Looks like CR #130 was also able to send huge messages.
 
Thank you!

1r0b1n0, 2017-11-19:
Hi,
The rosserial protocol encodes message length only on 16 bits (http://wiki.ros.org/rosserial/Overview/Protocol), so you have to modify both the server and client if you want to use messages bigger then 65kb.

For one of my projects I had to use bigger messages on Windows, so I made some changes to rosserial_server to make it support up to 32bit sized messages.
You can find my modifications to rosserial_server here : https://github.com/1r0b1n0/rosserial
I also developed a client-side implementation that supports 32bit : https://github.com/1r0b1n0/rosserial_qt ( but unlike rosserial_windows it depends on Qt )

julled, 2017-11-29:
Thank you @1r0b1n0 for your feedback!
I tried to use your code as it fits my problem, but i went into a issue, which i put on your rosserial_qt repository.