ros-drivers/rosserial #340: why not tf/transform_listener.h?

wbang8933, 2017-12-06:
I cann't find the tf/transform_listener.h file in ros_lib, why? If I want to start a tf listener,how can I do this?

romainreignier, 2017-12-07:
Rosserial is only a transport layer to allow a non-ROS system to communicate through a serial port or a TCP connection, you won't find any ROS method other than the Publisher, Subscriber, ServiceServer, ServiceClient...

mikepurvis, 2017-12-07:
Well, in fact there is a very rudimentary library for publishing transforms:

https://github.com/ros-drivers/rosserial/tree/jade-devel/rosserial_client/src/ros_lib/tf

But it's unlikely you'd actually want to publish a transform this way. The more conventional thing on a modern robot would be to publish a joint_state from the hardware, and then use robot_state_publisher to publish the appropriate transforms for your joints based on the URDF. When rosserial forks for Melodic, we may want to consider removing those tf-related headers from the client.

As far as _consuming_ TFs in a rosserial client, that is going to be tricky is only because it's a message with variable-length fields of other message types, which are awkward to work with in an environment without dynamic memory.

Is this for an actual microcontroller, or are you using rosserial as a bridge to Windows or some other system?

wbang8933, 2017-12-10:
I'm using it as a bridge to Windows